pvlib/pvlib-python #273: Errors in sapm() and singlediode()

gperonato, 2016-12-02:
I am getting these errors running the tmy_to_power notebook:
on sapm()
TypeError: sapm() takes 3 positional arguments but 6 were given

on singlediode()
ValueError: operands could not be broadcast together with shapes (21,) (8760,)

wholmgren, 2016-12-02:
This is related to #267. We did not update the tutorials for the syntax changes made going from pvlib 0.3 to 0.4. Check out the updated function calls here:

http://nbviewer.jupyter.org/github/wholmgren/pvlib-python/blob/tutorials041/docs/tutorials/pvsystem.ipynb

and here

http://pvlib-python.readthedocs.io/en/latest/api.html#pv-modeling

and see if you can make it work.

Even better if you make a pull request with the fixed code!

gperonato, 2016-12-02:
Thanks, it works now! I have made a pull request with the fixed code.

closed 2016-12-02 by gperonato