pvlib/pvlib-python #285: advice on modeling at high altitudes?

jforbess, 2016-12-16:
I am interested in modeling clear sky irradiance at different altitudes, including the stratosphere. My test run using my current clear sky setup with ineichen at an altitude of 20km produced an answer significantly greater than the dni_et I used. 

I'd appreciate any hints as to whether I need use something other than the default linke turbidities table, or whether there is a better algorithm for clear sky to use for high altitude modeling.

mikofski, 2017-01-15:
What about taking a ratio of the altitude to to the height of the atmosphere, apply that to precipitable water and AOD and use that either to calculate Linke or in Birds it simplified Solis.

EG: assume the sky is 100,000 meters, and your elevation is 5,000 [m]. So if the precip-water is 1.0[cm] at sea level, at 5km it's 0.95%.

Ditto for AOD. Then to calculate Linke, see #278, or just use either Bird or Solis with water and AOD directly.

mikofski, 2017-01-15:
BTW: this type of question might be good for the [pvlib-python Google group](https://groups.google.com/forum/m/#!forum/pvlib-python).

mikofski, 2017-01-15:
Sorry to hijack youre thread, but I've also set up a [pvlib-python slack channel](https://pvlib-python.slack.com). So far I've tried to invite @wholmgren @jforbess @bmeyers @zdefreitas @anomam and @adriesse.

cwhanse, 2017-01-16:
I found an empirical model for the attenuation of direct and global irradiance as a function of altitude here:
https://hal.archives-ouvertes.fr/hal-00470155/document

The formulas are hard to read (non-standard formatting) but the plots look promising.  I haven't used this model.

wholmgren, 2017-01-17:
See #297 for a place to discuss @mikofski's google group and slack channel.

stephenkaplan, 2017-01-19:
@mikofski can you please invite me to the slack channel? stephen.kaplan@firstsolar.com

wholmgren, 2017-02-15:
@jforbess I'm going to close this to clean up the issues board. I suppose people could continue to comment on it here, but perhaps the google group is a better place for this sort of thing.

closed 2017-02-15 by wholmgren