pvlib/pvlib-python #289: Default values for aoi modifier and temp model

stevenevados, 2017-01-11:
Comments inside pvsystem.py indicate that when the module files do not contain parameter values for either the Temp model or the AOI model that "default values" are used.

I do not see those default values anywhere and the current code gives an error when they are not found.  For both temp and AOI the code generates a "cannot infer model from input parameters" error in that case.

If needed I can create my own defaults, but this looks like an oversight.

wholmgren, 2017-01-13:
The AOI and temperature coefficients are keyword arguments in the function definitions in pvsystem.py. The pvsystem.PVSystem class methods respect the function defaults if your PVSystem object does not define them in its module_parameters attribute.

I don't think the problem is actually in pvsystem, though, because your error message sounds like a ModelChain initialization error. ModelChain tries to infer which models to use based on the coefficient names in the PVSystem's module_parameters attribute. However, the SAM module definitions do not include these parameters, so you must either add them to the module_parameters attribute or explicitly set some of the ModelChain methods.

That's my guess, but it's hard to know for sure without seeing a small reproducible example.

stevenevados, 2017-01-13:
The problem does turn out to be the lack of coefficients in the SAM modules
database.

I got around the problem by using coefficients for a similar module in the
Sandia Database.  The exact module I needed was not in the Sandia database,
but I think the numbers are close enough.

It seems to me that ModelChain should revert to default values for the
temperature and aoi parameters when they are not available from the
database.  In line comments suggest that should happen, but in fact if the
database parameter set is missing the relevant numbers, the code issues a
"cannot infer [aoi|temp] model" and dies.

Steve Ihnen
Principal Solar Energy Scientist
(425) 877-9451

wholmgren, 2017-01-13:
Which in line comments are you referring to?

The problem is that ModelChain doesn't have default models -- not that the models themselves don't have default parameters. You can explicitly tell the ModelChain which models to use by adding ``aoi_model='physical', temp_model='sapm'`` to your ModelChain creation line. The associated PVSystem object will then end up using the default model parameters. For example, this code works despite the fact that the PVSystem object does not have a module or inverter specification.

```python
pvsystem = PVSystem()
location = Location(32.2, -110.9)
mc = ModelChain(pvsystem, location, orientation_strategy=None,
                dc_model='pvwatts', ac_model='pvwatts',
                aoi_model='physical', spectral_model='no_loss',
                temp_model='sapm', losses_model='no_loss')
```

Of course, it will fail at when you try to run ``mc.run_model(times)`` because the dc models need some kind of module information to work.

I'm sure that this behavior could be documented better, and I'd welcome a pull request to do so.

cwhanse, 2017-01-16:
I'm in favor of pvlib gracefully handling missing, or unspecified, data, if accompanied by a message of some kind that's hard to miss, which informs the user of the situation.  One purpose of pvlib is to promote transparency and rigor in pv modeling, and in my opinion, use of default values should be a conscious decision by the modeler.

wholmgren, 2017-02-15:
This seems like a dead non-issue, but feel free to reopen and elaborate if you think otherwise.

closed 2017-02-15 by wholmgren