pvlib/pvlib-python #291: Simple analytical Azimuth method

mikofski, 2017-01-14:
From Duffie/Beckman, Seinfeld/Pandis, etc. 1-line, very accurate.

wholmgren, 2017-01-14:
What's the application? Efficient calculation of solar noon? We already have too many ways to calculation solar position, so I'm reluctant to add more unless there's a good use case for them.

mikofski, 2017-01-14:
We have used it in the past, so I have to continue to support those calculations. Apples to apples.

It's literally a one liner, so not sure what the concern is. I think of pvlib as not just a tool box but also a single source of truth for PV formulas. Having this 1-liner completes the solar position calculation. Any IE coming across a random equation will need to vet it, whereas _if the analysis uses **pvlib**_, then the vetting has already been done. This saves time and money.

wholmgren, 2017-01-14:
Now that I've looked at your updated #276 I think I understand what you're proposing. You want the equivalent of ``def solar_zenith_analytical(latitude, hour_angle, declination):`` for azimuth?

adriesse, 2017-01-14:
Just curious, what line are you referring to?

mikofski, 2017-01-15:
Yes, thx. I'll probably submit a PR early next week 😀

wholmgren, 2017-08-07:
closed by #349

closed 2017-08-07 by wholmgren