pvlib/pvlib-python #294: change PVSystem default module_parameters and inverter_parameters to empty dict

wholmgren, 2017-01-14:
``PVSystem`` currently sets the default ``module_parameters`` and ``inverter_parameters`` to ``None``. I think we should change the default values to be empty dicts. Since we want unique dicts for each object, the ``__init__`` method signature would remain the same but we'd add the code below to the method.

```python
if modules_parameters is None:
    self.module_parameters = {}
else:
    self.module_parameters = module_parameters
```

...and similar for ``inverter_parameters``.

This has two benefits:

1. It allows a couple of ``PVSystem`` methods to fall back to default values when module_parameters has not been specified. (this problem could also be resolved by making the ``tools._build_kwargs`` function yield ``{}`` from ``None`` input.)
2. It allows users to set new properties without needing to create a new dict to start with. e.g.

 ```python
mypvsystem = PVSystem()
mypvsystem.module_parameters['pdc0'] = 0.05
``` 

I think this is a small enough api change that it could go in the 0.4.4 release.

mikofski, 2017-01-14:
There is some pep8 rule about allowing mutable defaults in function signatures, I'll try to find it. But whenever I try to set an empty dict as an argument default I get that warning from pylint, pyflakes, pep8, etc.

Instead you could add a line that if the arg is `None` then set it to an empty dict inside the function def.

Note when modules are imported the function and class definitions are read, but not their contents. Maybe that's why; having a lot of mutables that can basically never go out of scope would take up resources and make reference counting tricky. Not to mention that having reference that pass through functions can cause issues too, IE: basically another method could change the contents of that reference (from empty `dict` to something else) then that would be tricky to troubleshoot/debug.

But the contents inside a function def are only read when the function is called, and for the most part any reference inside the function is trashed when the function returns and it's references go out of scope. Only globals and argument references survive, which is why they're tricky.

wholmgren, 2017-01-14:
that's what I proposed above...

mikofski, 2017-01-14:
Oops, sorry 😶

PR #304 (2017-02-14): set default module and inverter_parameters to empty dict
closes #294. This is a very minor API change, but let me know if you think it should be delayed until 0.5.
closed 2017-02-14 by wholmgren