pvlib/pvlib-python #296: build and serve docs using ghpages in addition to rtfd.org

mikofski, 2017-01-17:
Priority: Very very low

Since there seem to be occasional gliches and errors serving docs at [rtfd.org](rtfd.org) maybe the ghpages at pvlib.github.io/pvlib-python should be activated. I know it's an additional PITA, but  maybe there are also upsides like a chance to test rendered docs, an alternate site in case rtfd goes down? [Maybe there is a way to automate build and deploy using travis?](https://gist.github.com/domenic/ec8b0fc8ab45f39403dd). [It seems like there are a few examples?](https://gist.github.com/domenic/ec8b0fc8ab45f39403dd) [Here's a blog post](http://www.steveklabnik.com/automatically_update_github_pages_with_travis_example/) and [an article in medium](https://medium.com/@nthgergo/publishing-gh-pages-with-travis-ci-53a8270e87db#.b75zbk2g8).

Anyway, just an idea, based on the [issue with forecasting docs described in this SO post](http://stackoverflow.com/a/41686297/1020470).

wholmgren, 2017-01-17:
I have considered moving the documentation to gh-pages in the past. It seems like a minor pain to make it work for the master version, but branching, releases, and forks look more complicated. doctr looks helpful for some of that. I'm not happy with either ghpages or rtd at this point.

I recently watched the metpy project refactor its documentation Unidata/MetPy#283 and I think there's some good examples for us there.

wholmgren, 2020-09-09:
Readthedocs has really improved over the last few years so I no longer see any benefit to this. I'm not opposed to adding strict doc builds to the CI to catch errors/warnings, but let's leave the complications of serving docs to RTD.

closed 2020-09-09 by wholmgren