pvlib/pvlib-python #297: pvlib python communication forums

wholmgren, 2017-01-17:
What do people think about these new communication forums?

- pvlib tag on stackoverflow: http://stackoverflow.com/questions/tagged/pvlib
- pvlib python google group: https://groups.google.com/forum/m/#!forum/pvlib-python
- pvlib python slack organization: https://pvlib-python.slack.com/

Here are my thoughts:

First, all credit goes to @mikofski for pushing this forward and setting up some things for us to try! Thanks!

I am concerned that too many forums could be counter productive, but I think we're big enough now that we could benefit from more than GitHub issues. Watching the project on GitHub does allow people to stay up to date on every little thing in the pvlib python world, but that is way too much information for many people. Spreading the communication over multiple platforms lets people focus on what they really care about at the cost of making things more complicated and potentially diluting the viewership of any given item.

I think it would be nice if we would use GitHub for bug tracking and development discussions, people would ask usage questions on Stack Overflow, and we’d use a mailing list for new version announcements. 

I know that a few people have tried to ask pvlib questions on stack overflow in the past with varying degrees of success. I am going to start watching Mark's new pvlib tag. I don't have much experience with google groups, but at this point I prefer stackoverflow for usage questions. In any case, while we're getting started with stack overflow and/or google groups, I think that people should feel free to open and then close a GitHub issue that links to the new post.

On a related note, [Sibbell](https://sibbell.com/about/) is a nice tool that will send you an email when there are new releases of any of your starred repositories. 

I like Slack for quick, informal conversations in which I don’t have to stress about if I’m going to come across the wrong way in a public forum. I don't like the constant din of interruptions on Slack, though.

I also get requests for help via email from time to time. I don't like answering questions via email because only a handful of people have an opportunity to learn from the exchange.

I don’t know what the ideal approach is, but I think we should try something new and see if it catches on!

Finally, we should add some notes to the documentation once we come to a consensus.

jforbess, 2017-01-17:
>    questions/tagged/pvlib
>    forum/m/#!forum/pvlib-python

I like stack overflow for usage questions, despite some of the community's
lack of understanding of the timezone issue I tried to ask about. :)  "Just
use UTC" doesn't work when you're modeling data that was labeled in one
timezone that doesn't match the physical timezone you need to use to create
solarposition. I will also start watching the new tag.

I agree it is nice to avoid direct email to keep information available for
more people to use.

I do not have a slack account for the fear that the constant interruptions
will keep me from accomplishing anything. But I hear it's what the cool
kids use. But that will probably change next year, because it always does.

Jessica

cwhanse, 2017-01-17:
This:

> I think it would be nice if we would use GitHub for bug tracking and development discussions, people would ask usage questions on Stack Overflow, and we’d use a mailing list for new version announcements.

I've never used Slack, ergo I'm not cool.  I don't think I would except for a project that needed near-real time coordination.  I don't think that's needed for pvlib development, and I wouldn't participate in that manner regarding usage - I don't want to default into a free 'help desk'.

For Matlab pvlib we get many inquiries but only by email.  In hindsight, we would be better served if we'd taken the time to redirect people to stack overflow, I think.

mikofski, 2017-01-17:
FYI: the google group site for desktop is without the `/m/`
https://groups.google.com/forum/#!forum/pvlib-python

adriesse, 2017-01-18:
Email is by far my most-used form of communication and I still like the old list-serv and newsgroup methods for  broader communication.  (How's that for un-cool!)  Having said that, I've grown to like at least some of the features of stackoverflow and github, so who knows what's next for me.

I haven't felt a strong need for additional communication platforms around pvlib, but I'm curious to see what people will gravitate toward.

closed 2017-02-17 by wholmgren

najiya-12, 2022-05-16:
@wholmgren could you please send me invitation to join the pvlib-python.slack.com! My email is Najiya.m.omar@gmail.com. I was trying to join the slack, but it's asked me to have invitation. 
Thank you

wholmgren, 2022-05-16:
As far as I know, the channel is open. I don't know how to manage slack and I no longer use it. Perhaps we should shut it down.

cwhanse, 2022-05-16:
> Perhaps we should shut it down.

I agree. There's essentially no traffic in that channel.

mikofski, 2022-05-16:
This is it, do we have like a warning or just cut the cord?
[image]

mikofski, 2022-05-16:
Hmm maybe we keep it to avoid spammers from stealing it? It doesn't hurt to leave it there, but maybe it's messy? idk what to do

PR #312 (2017-02-17): minor documentation updates
A handful of minor updates to the documentation. Closes #297 (communication forums).
Partially addresses #302 (LT/AOD examples). Adds some notes about recommended solar position methods (relevant to #301). Updates the doc environment to use newer packages.

It's rendered here:

http://wholmgren-pvlib-python-new.readthedocs.io/en/docupdates/index.html

Let know if you have any suggestions.