pvlib/pvlib-python #299: Forecasting Docs build error

nparley, 2017-01-18:
The docs for forecasting need updating http://pvlib-python.readthedocs.io/en/latest/forecasts.html as there are many build errors, Maybe all caused by `netCDF4` import error at the top?

wholmgren, 2017-01-18:
Yep. https://github.com/pvlib/pvlib-python/pull/180#issuecomment-235935887

wholmgren, 2017-01-18:
An example of how it fails: https://readthedocs.org/projects/wholmgren-pvlib-python-new/builds/4753267/

wholmgren, 2017-01-18:
Just tried it again and it works now!

https://readthedocs.org/projects/wholmgren-pvlib-python-new/builds/4913582/

Well, it fails with the known sqllite error and has some other warnings. But the docs still build:

http://wholmgren-pvlib-python-new.readthedocs.io/en/rtdnc/forecasts.html

I will clean up a few things and make a pull request soon.

PR #300 (2017-01-19): enable forecasting doc page build on readthedocs
closes #299 

http://wholmgren-pvlib-python-new.readthedocs.io/en/rtdnc/forecasts.html
closed 2017-01-19 by wholmgren