pvlib/pvlib-python #302: add discussion and examples of Angstrom and Kasten formulas to docs

mikofski, 2017-02-04:
Related to issue #269 and a follow on to PR #278 which adds the Kasten pyrheliometric formula to calculate Linke turbidity factor given broadband AOD and precipitable water, the Bird-Hulstrom broadband AOD correlation and Angstrom formulas to calculate alpha exponent from AOD at two wavelengths and to calculate AOD given one measurement and alpha.

This issue proposes to add a brief discussion and some examples to the clear sky documentation. The discussion can also mention other methods not currently included in pvlib-python.

mikofski, 2018-08-31:
Any comments on how to approach this? How long of a discussion? What's the main goal? I'd like to close this out, so any pointers would help a lot. thx

Here are some references with links to more:
* [Use of Measured Aerosol Optical Depth and Precipitable Water to Model Clear Sky Irradiance](https://www.researchgate.net/publication/318470239_Use_of_Measured_Aerosol_Optical_Depth_and_Precipitable_Water_to_Model_Clear_Sky_Irradiance)
* [Evaluation and Correction of the Impact of Spectral Variation of Irradiance on PV Performance](https://www.researchgate.net/publication/309648716_Evaluation_and_Correction_of_the_Impact_of_Spectral_Variation_of_Irradiance_on_PV_Performance)

wholmgren, 2018-08-31:
A paragraph or two in clearsky.rst. Only enough to give people a sense of what’s possible.

PR #563 (2018-09-07): DOC: Discuss Angstrom Kasten TL, set default alpha
pvlib python pull request guidelines
====================================

Thank you for your contribution to pvlib python! You may delete all of these instructions except for the list below.

You may submit a pull request with your code at any stage of completion.

The following items must be addressed before the code can be merged. Please don't hesitate to ask for help if you're unsure of how to accomplish any of the items below:

 - [x] Closes #302
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en [... cut]
commit ec329e28 (2018-09-07): DOC: Discuss Angstrom Kasten TL, set default alpha (#563)

* DOC: add example for Kasten pyrheliometric formula

* closes #302
* add references to Kasten 96 and Molineaux
* discuss using Kasten and calculating broadband AOD usinge Angstrom turbidity model

* DOC: update what's new, add refs to Bird & Hulstrom and Angstrom

* DOC: split sentences, replace abbrev. AOD

* use TMY in pvlib/data, lowercase MBARS, use non-specific variable names
* split output from readtmy3 instead of using indexing for clarity

* DOC: fix up ipython example

* DOC: split sentences, revise, add angstrom example

* E [... cut]
closed 2018-09-07 by wholmgren