pvlib/pvlib-python #305: gaussian process regression to estimate PV performance

cbirkj, 2017-02-06:
Historical system and weather data can be used to develop PV models.  In the past, least-squares regression has been used to model systems based on past data.  However, the Gaussian Process Regression (GPR) algorithm can provide a more accurate estimate of PV performance. 
[image]
The algorithm can estimate power, current, and voltage with a smaller % error than the least-squares approach: 
[image]
The above figure shows the percent error for estimated power based on three different training data sets: 1) complete set, 2) binned, and 3) random.  The GPR reached 5% while the least-squares regression did not drop below 8%.

There are other Python packages that provide GPR algorithms.  However, it may be difficult for people that do not have a machine learning background to implement.  I propose to provide a GPR algorithm that is customized for PV applications.

**Do you think that it would be valuable to add GPR capabilities to the existing PVLIB tool set?**

Here is sample code for the GPR function:

def kernel(sigma,alpha,l,x1,x2):
	
        '''
	Rational Quadratic Kernel
	:param: sigma (sigma^2 determines average difference of function from mean)
	:param: alpha (determines relative weighting of variations)
	:param: l (length scale)
	:param: x1 - training set input
	:param: x2 - teting set input
	
	:return: kernel
	'''
	s = distance.cdist(x1,x2,'euclidean')
	k = (sigma**2) * (1+((s)**2/(2*alpha*l**2)))**-alpha
	return k
		
def gp(x1,y1,x2,sigf,alpha,l,sign=0.1):
    
    K = kernel(sigf,alpha,l,x1,x1)				
    Kss = kernel(sigf,alpha,l,x2,x2)
    Ks = kernel(sigf,alpha,l,x1,x2)
    Ks = np.transpose(Ks)	
    K = K + np.identity(len(K)) * noise								# Add noise
    L = np.linalg.cholesky(K)  
    beta = np.linalg.solve(L.transpose(), np.linalg.solve(L,y1))
    Mu = np.array([np.dot(Ks,beta)]).T
    
    d = np.dot(Ks,inv(K))
    m = np.dot(d,np.transpose(Ks))
    s = Kss - m
    di = np.diag(s)
    Sigma = np.transpose(np.array([np.sqrt(di)]))
  
    return Mu,Sigma

cwhanse, 2017-02-06:
Birk,

While I’m not opposed to making it easier to combine pvlib with other python capabilities, what you are proposing amounts to an alternative PV system model.  It’s not clear there’s any advantage to your approach over well-fit PVsyst or other models.  I think the discussion about the need for, and value of, the GPR approach should take place, but not here.  Mark, Will, perhaps we can cross-post this to one of the forums that were linked a few weeks back?

Cliff

[graph_model_power]<https://cloud.githubusercontent.com/assets/5728325/22664287/32889cb6-ec6d-11e6-9e76-5e929d00df66.png>
[graph_mape_power]<https://cloud.githubusercontent.com/assets/5728325/22664362/99b1adba-ec6d-11e6-9178-f274e44024cb.png>

wholmgren, 2017-02-06:
Birk asked me via email what I thought about it and I suggested that he propose it to the community here. I think some discussion of the merits here is fine. However, I agree that a more involved discussion might be better suited to the google forum:

https://groups.google.com/forum/#!forum/pvlib-python

I'm not sure where to draw the line between "some discussion" and "a more involved discussion" but let's go to the google forum for a change of pace.

Birk, can also you link to your papers/talks/posters when you repost it to the google forum? Thanks!

adriesse, 2017-02-06:
I don't mind seeing this here--I'd have missed it on google.  

Is there more than one question here?  One seems to be about the merits of the technique, and the other about whether or not its implementation should be included in pvlib-python.

thunderfish24, 2017-09-13:
@cbirkj

Hi Birk,

  This sounds very appealing. Can you provide one good PV+GPR literature reference to help me come up to speed?

  Generally, I don't mind custom "wrappers" for specific applications, but I do think we should be mindful about committing pvlib to maintaining custom GPR code.

-Mark Campanelli