pvlib/pvlib-python #306: Loss Factor Model

cbirkj, 2017-02-06:
The Loss Factor Model (LFM) is used to analyze current and voltage (I-V) curves and model system performance ([Stein et al](http://energy.sandia.gov/wp-content//gallery/uploads/4CO-11-1a.pdf)).  

Similar to my last issue post, I am wondering what people's thoughts are on adding LFM parameter estimation to the PVLIB library?

Birk

adriesse, 2017-02-06:
Seems like a good idea to me.  There are lots of papers on it.

steve-ransome, 2017-02-25:
Hi all
Just to let you know I am working on adding the LFM to PV LIB (also with Juergen Sutterlueti from Gantner Instruments) and hope to have something in a couple of weeks.

adriesse, 2017-02-26:
Good work!  Python? or Matlab?

steve-ransome, 2017-02-26:
Python.

cwhanse, 2017-08-24:
Hi Steve, any progress on this?

wholmgren, 2020-08-26:
#925 contains the recent discussion of this topic.

closed 2020-08-26 by wholmgren