pvlib/pvlib-python #307: Daylight saving timestamp issue in solarposition.ephemeris

anomam, 2017-02-07:
In solarposition.ephemeris:
```
    # make output DataFrame
    DFOut = pd.DataFrame(index=time)
    DFOut['apparent_elevation'] = ApparentSunEl
    DFOut['elevation'] = SunEl
    DFOut['azimuth'] = SunAz
    DFOut['apparent_zenith'] = 90 - ApparentSunEl
    DFOut['zenith'] = 90 - SunEl
    DFOut['solar_time'] = SolarTime
```
I get an error when assigning ``ApparentSunEl`` to the ``DFOut`` column:
```
ValueError: cannot reindex from a duplicate axis
```
and I believe the reason comes from the fact that DFOut uses ``time`` as index and ``ApparentSunEl`` uses ``time_utc`` as index; and I think it raises an error because in my case ``time`` uses daylight saving and contains duplicate indices. Using ``ApparentSunEl.values`` instead should solve the issue.

wholmgren, 2017-02-07:
I'm guessing that the same thing happens with ``solarposition.pyephem``. I don't think it should happen with ``solarposition.spa_python``. We'll probably need ``.values`` on all of the columns. 

It would be good to add a test of each method that uses duplicated labels due to DST. You could copy this test

https://github.com/pvlib/pvlib-python/blob/master/pvlib/test/test_solarposition.py#L321-L342

but use a 2 element ``times`` index that occurs on the hour before and the hour of DST. You'd also copy the fixture ``expected_solpos_multi`` to create ``expected_solpos_dst_transition`` or something like that.

cwhanse, 2018-05-07:
Old issue, but in my opinion requires no change to ephemeris. Times in data should be localized and daylight savings conflicts resolved by the user prior to this function.  Maybe this is a requirement for the future pvlib.io.

I recommend closing.

wholmgren, 2018-08-14:
This might be fixed by #529. I'm going to close for now but please reopen if you continue seeing the issue and plan to make a pull request with a fix.

closed 2018-08-14 by wholmgren