pvlib/pvlib-python #320: Citing the module and inverter database

uvchik, 2017-04-10:
Is there a way to cite the module and inverter database? 

I cannot find an official link or some meta information about these files.

May anybody help?

cwhanse, 2017-04-10:
I’d cite the System Advisor Model Version 2015.6.30 as the source for all three .csv tables.  However, because the NREL SAM team didn’t originate any of the data in those tables, I’d word the citation to reflect that fact:

‘Sourced from System Advisor Model Version 2015.6.30, downloaded from https://sam.nrel.gov/’

The California Energy Commission is the source for the CEC inverter and module data, but I don’t believe that the CEC distributes that content itself.  I know that Sandia doesn’t reliably share the Sandia module table; we send (very infrequent) updates to NREL for distribution with SAM.

Cliff

uvchik, 2017-04-21:
Thanks a lot. Following your information I could find the inverters on the gosolarcalifornia website:
http://www.gosolarcalifornia.org/equipment/inverter_tests/summaries/

Furthermore I found an overview about the libraries on https://sam.nrel.gov/libraries . There are at least some information about where each library comes from.

I guess that is all I will find.

closed 2017-04-21 by uvchik