pvlib/pvlib-python #321: PV Monitoring

cbirkj, 2017-04-28:
PV_LIB has packages for system modeling and forecasting.  What does the group think about a monitoring package? 

I have some python code that collects current, voltage, and temperatures from a Modbus device.  Currently, the code is running successfully on a raspberry pi and collecting data at 1 second intervals and storing the data in a MySQL database.

What do you think?

thunderfish24, 2017-04-28:
Is the idea of the package to help ensure that monitoring data can be integrated with other PV_LIB tools?

wholmgren, 2017-04-28:
Would pecos be a better fit?

cbirkj, 2017-04-28:
Good point.  Pecos could be a better fit.

djaberpy, 2017-04-28:
That great. Which kind of pv panel.

wholmgren, 2017-05-10:
@cbirkj I saw a pecos pull request with similar functionality. can this issue be closed now?

closed 2017-05-11 by cbirkj