pvlib/pvlib-python #322: replace all instances of .ix

wholmgren, 2017-05-04:
Pandas 0.20 deprecates the ``.ix`` indexer and we need to make a handful of minor changes to stay current.

https://github.com/pvlib/pvlib-python/search?utf8=✓&q=.ix&type=

http://pandas-docs.github.io/pandas-docs-travis/whatsnew.html#whatsnew-0200-api-breaking-deprecate-ix

I'd welcome a PR for this.

morrme, 2017-05-04:
What do you want to change the references to: `.loc `or `.iloc` ?

cwhanse, 2017-05-05:
Depends whether we want position or label indexing.  The problem with .ix is that it tried to handle both cases.

adriesse, 2017-05-05:
Are these methods ok?

```
df.column_name.[row_number]
df.column_name.iloc[row_number]
df.iloc[row_number].column_name

```

wholmgren, 2017-05-05:
In ``timetimezones.rst``, ``tmy3_data.ix[0:3, ['GHI', 'DNI', 'AOD']]`` becomes ``tmy3_data[tmy3_data.index[0:3], 'GHI', 'DNI', 'AOD']``

In ``forecast.py``, I think you can simply use ``[self.output_variables]``.

I think the case in ``irradiance.py`` will work with ``loc``.

PR #328 (2017-05-30): remove all instances of .ix
closes #322.

see https://pandas.pydata.org/pandas-docs/stable/whatsnew.html#deprecate-ix
closed 2017-05-30 by wholmgren