pvlib/pvlib-python #334: pvlib.pvsystem PV curve

evgenyna, 2017-06-04:
Hi.
Please advise why i'm gettitng the error
File "C:\Users\***\Anaconda3\lib\site-packages\pvlib\pvsystem.py", line 1938, in i_from_v
    np.exp(Rsh*(Rs*(IL+I0)+V) / (nNsVth*(Rs+Rsh))) /
TypeError: must be str, not int
Thank you in advance, Evgeny.
`import pandas as pd
import numpy as np
import matplotlib.pyplot as plt
from pvlib.pvsystem import singlediode, v_from_i, i_from_v, retrieve_sam

def fivepoints_to_frame(pnt):
    """
    Converts a 1 dimensional, dict-like singlediode or sapm result
    to a 5 row DataFrame with columns current and voltage.
    Users can iterate over the rows of a multidimensional
    singlediode or sapm result, if necessary.
    """
    ivframe = {'i_sc': (pnt['i_sc'], 0),
               'p_mp': (pnt['i_mp'], pnt['v_mp']),
               'i_x': (pnt['i_x'], 0.5*pnt['v_oc']),
               'i_xx': (pnt['i_xx'], 0.5*(pnt['v_oc']+pnt['v_mp'])),
               'v_oc': (0, pnt['v_oc'])}
    ivframe = pd.DataFrame(ivframe, index=['current', 'voltage']).T
    ivframe = ivframe.sort_values(by='voltage')

    return ivframe

resistance_shunt = 16
resistance_series = 0.094
nNsVth = 0.473
saturation_current = 1.943e-09
photocurrent = 7
module_parameters = retrieve_sam('cecmod')['Example_Module']

v_oc = v_from_i(resistance_shunt, resistance_series, nNsVth, 0, saturation_current, photocurrent)
voltage = np.linspace(0, v_oc, 100)

current = i_from_v(resistance_shunt, resistance_series, nNsVth, voltage, saturation_current, photocurrent)

fivepnts = singlediode(
    module_parameters, photocurrent, saturation_current, resistance_series, resistance_shunt, nNsVth)
ivframe = fivepoints_to_frame(fivepnts)

fig, ax = plt.subplots()
ax.plot(voltage, current)
ax.scatter(ivframe['voltage'], ivframe['current'], c='k', s=36, zorder=10)
ax.set_xlim(0, None)
ax.set_ylim(0, None)
ax.set_ylabel('current (A)')
ax.set_xlabel('voltage (V)')`

wholmgren, 2017-06-04:
We removed the unnecessary ``module_parameters`` argument from the [``singlediode``](http://pvlib-python.readthedocs.io/en/latest/generated/pvlib.pvsystem.singlediode.html#pvlib.pvsystem.singlediode) function about a year ago.

closed 2017-06-04 by wholmgren

evgenyna, 2017-06-04:
Thnk you