pvlib/pvlib-python #335: Documentation improvement

birgits, 2017-06-08:
I just stumbled across two things in the documentation where it would be helpful to correct them:

* For the `surface_tilt` in the `PVSystem` class it says: 

> Up=0, horizon=90.

This is a bit confusing since I guess it is meant as defined in the `basic_chain`:

> The tilt angle is defined as degrees from horizontal (e.g. surface facing up = 0, surface facing horizon = 90)

Is that right? Maybe we could use the same definition in the `PVSystem` class.

* In the docstring of the `Modelchain` class the default value for the `orientation_strategy` `south_at_latitude_tilt` is missing.

wholmgren, 2017-06-08:
> Is that right?

That is right. It could be updated in a lot of places. Many other parameters could be made more consistent.

> In the docstring of the Modelchain class the default value for the orientation_strategy south_at_latitude_tilt is missing.

We generally have not put default values in the doc strings. I think they're useful, though, so feel free to add this. I'd only ask that we add defaults to all parameters of a function, instead of only a single parameter. See also #290.

uvchik, 2017-06-12:
According to #290 we may want to add a comment to the docstring entry of orientation_strategy:

`(default: 'south_at_latitude_tilt') WARNING: The default will be set to None from version 0.5.x on.`

```diff
def __init__(self, system, location,
-            orientation_strategy='south_at_latitude_tilt',
+            orientation_strategy='old_default`,
             clearsky_model='ineichen',
             ....

+     if orientation_strategy == 'old_default'
+        orientation_strategy='south_at_latitude_tilt'
+        warnings.warn("Default value will be changed.... ", FutureWarning)
```

People can switch of this warning by defining the parameter explicitly and so they ready for the changes.

wholmgren, 2017-08-01:
closed by #336

closed 2017-08-01 by wholmgren