pvlib/pvlib-python #338: Set negative values of AOI loss to 0 rather than np.nan

cwhanse, 2017-06-13:
The current AOI loss models set negative values to np.nan, rather than zero.  When np.nans are propagated to the effective irradiance model, the beam irradiance (and the effective irradiance) becomes np.nan rather than 0.  As a consequence, pvlib cannot compute effective irradiance when AOI>90, i.e., when the sun is behind the array plane.  In this circumstance there's still diffuse irradiance on the array plane.

I think it would be better to implement the AOI loss models as

f(AOI) = model value if AOI<90
f(AOI) = 0 otherwise.

jkfm, 2017-06-13:
Thanks Cliff for opening an issue! It refers to my question in the [google group](https://groups.google.com/forum/#!topic/pvlib-python/AiWA6v4x3GM).

If I find time in the next couple of days, and if there is consent on the issue, I will start working on a pull request.

cwhanse, 2017-06-13:
(waves at Johannes)  you’re welcome ☺

wholmgren, 2017-06-13:
I can see arguments for nans or 0s. I think nan could be the right result because maybe it doesn't make sense to talk about an AOI modifier when the sun is behind the array plane. In a similar fashion, pvlib's relative airmass function returns nan when the zenith angle is greater than 90. 

On the other hand, I can think of at least a few of my results where I've failed to account for these nans, so that's a good indicator that something should change.

One thing is for certain: ModelChain fails to account for the nans in its effective_irradiance calculation and that should be fixed as soon as possible.

wholmgren, 2017-06-14:
The code below should allow you to monkey patch your modelchain objects until we get this fixed:

```python
import types
def effective_irradiance_model_corrected(self):
    fd = self.system.module_parameters.get('FD', 1.)
    self.effective_irradiance = self.spectral_modifier * (
        self.total_irrad['poa_direct']*self.aoi_modifier.fillna(0) +
        fd*self.total_irrad['poa_diffuse'])
    return self

mc = ModelChain(system, location, **your_normal_kwargs)
mc.effective_irradiance_model = types.MethodType(effective_irradiance_model_corrected, mc)
mc.run_model(times, weather=weather)  # as if nothing has changed
```

AC power without the patch:

[image]

and with the patch:

[image]

jkfm, 2017-06-14:
Thanks for the fix, @wholmgren. I forked the project and changed the functions for myself though (see PR).

wholmgren, 2017-08-01:
closed by #339

closed 2017-08-01 by wholmgren