pvlib/pvlib-python #347: sapm takes 3 positional arguments but 6 were given

AmyneTRABELSI, 2017-07-05:
dc = pvlib.pvsystem.sapm(module, total_irrad['poa_direct'],total_irrad['poa_diffuse'], temps['temp_cell'],am_abs, aoi) 

TypeError: sapm() takes 3 positional arguments but 6 were given

wholmgren, 2017-07-05:
The ``sapm`` function signature changed about a year ago. http://pvlib-python.readthedocs.io/en/latest/generated/pvlib.pvsystem.sapm.html#pvlib.pvsystem.sapm

closed 2017-07-24 by wholmgren

mikofski, 2017-09-01:
You can still use [`sapm_effective_irradiance`](http://pvlib-python.readthedocs.io/en/latest/generated/pvlib.pvsystem.sapm_effective_irradiance.html?highlight=sapm_effective_irradiance) as in the [docs package overview](http://pvlib-python.readthedocs.io/en/latest/package_overview.html#procedural) to use the legacy api:

    effective_irradiance = pvlib.pvsystem.sapm_effective_irradiance(
        total_irrad['poa_direct'], total_irrad['poa_diffuse'],
        am_abs, aoi, module)