pvlib/pvlib-python #352: `repr()` on a `ModelChain` instance fails if `orientation_strategy = 'None'`

jgaffiot, 2017-07-25:
If `orientation_strategy` is set to `'None'` when creating a `ModelChain`, the attribute `self._orientation_strategy` is set to `None` in the orientation_strategy setter (from line 353).
In the `__repr__` method, just above, each attribute is printed thanks to the inner function `getmcattr(self, attr)`. And if the argument `attr` is `None`, then the function returns `None`.
Then, when building the returned chain, `None` is passed to the concatenation operator, which fails.

jgaffiot, 2017-07-25:
Can be fixed by changing line 344 to:
    
     ('{}: {}'.format(attr, getmcattr(self, attr)) for attr in attrs)))

I would have done the pull request if I had the rights.

wholmgren, 2017-07-25:
Thanks, it would be great if you could submit a pull request with the fix.

wholmgren, 2017-07-25:
``PVSystem`` and ``SingleAxisTracker`` ``__repr__`` methods enclose ``getattr(self, attr)`` in the ``str`` function rather than using ``format``:

https://github.com/pvlib/pvlib-python/blob/master/pvlib/pvsystem.py#L159
https://github.com/pvlib/pvlib-python/blob/master/pvlib/tracking.py#L62

Perhaps we should use the same approach in all three ``__repr__`` methods. It's not clear to me if one is better than the other.

jgaffiot, 2017-07-25:
Hi wholmgren,

I'm unfamiliar with the pull request system in third-party project (even with mine, I'm still learning GitHub).
I naïvely tried:

    git push origin fix_repr_model_chain

but of course I'm not able to access the `pvlib` repository.

Concerning the fix, I think that using `format` is slightly more clean and pythonic than concatenation with '+', and is less error prone. I would use `format` for the 3 classes.

wholmgren, 2017-07-25:
I am happy to help new contributors get oriented here. Start with this section of our documentation and then let me know if you have further questions:

http://pvlib-python.readthedocs.io/en/latest/contributing.html#how-to-contribute-new-code

jgaffiot, 2017-07-25:
I have submitted a pull request (#353) to fix the bug.

closed 2017-07-26 by wholmgren