pvlib/pvlib-python #354: add PULL_REQUEST_TEMPLATE.md file

wholmgren, 2017-07-25:
I'd like to make a ``.github`` directory with a ``PULL_REQUEST_TEMPLATE.md`` file. The XArray/Pandas template would work for me:

```
 - [ ] Closes #xxxx
 - [ ] Tests added / passed
 - [ ] Passes ``git diff upstream/master | flake8 --diff``
 - [ ] Fully documented, including `whats-new.rst` for all changes and `api.rst` for new API
```

https://help.github.com/articles/creating-a-pull-request-template-for-your-repository/

Later, we might consider adding a CONTRIBUTING.md, ISSUE_TEMPLATE.md, or CODE_OF_CONDUCT.md files to the ``.github`` directory. I'll leave those to separate issues.

thunderfish24, 2017-07-26:
If pvlib were >=1.0.0, then I would suggest a [SemVer-related](http://semver.org/) checklist. I'm not sure if we can capture the minor vs. patch guidelines at this point in pre-1.0.0 releases.

wholmgren, 2017-08-11:
closed by #355

closed 2017-08-11 by wholmgren