pvlib/pvlib-python #358: Get a peak AC power around 6:00 am in the morning

jielianguo, 2017-08-04:
I run the bellowing code and got a high AC power at 6:00 am.
  
```python
import pandas as pd
import matplotlib.pyplot as plt
import datetime
import sys
# seaborn makes the plots look nicer
import seaborn as sns; sns.set_color_codes()
from pvlib.forecast import GFS, NAM, NDFD, HRRR, RAP,ForecastModel
from pvlib.pvsystem import PVSystem, retrieve_sam
from pvlib.tracking import SingleAxisTracker
from pvlib.modelchain import ModelChain
from pvlib.location import Location

sandia_modules = retrieve_sam('sandiamod')
cec_modules = retrieve_sam('CECMod')
cec_inverters = retrieve_sam('cecinverter')
module = sandia_modules['Trina_TSM_240PA05__2013_']
inverter = cec_inverters['SolarEdge_Technologies__SE9K_208V__CEC_2013_']
print(module)

latitude, longitude, tz = 53.622537, -113.644237, 'Canada/Mountain'
start = pd.Timestamp('07/30/2017 00:00:00', tz=tz)
end = start + pd.Timedelta(days=5)
fx_model = RAP(set_type ='best') 
fx_data = fx_model.get_processed_data(latitude, longitude, start, end)

system = PVSystem(module_parameters=alberco_module_similar,\
                  inverter_parameters=alberco_inverter,\
                  modules_per_string=22,\
                  strings_per_inverter=2,surface_tilt = 10,surface_azimuth =150 )
mc = ModelChain(system, fx_model.location,name= 'pv', losses_model='no_loss')
mc.run_model(fx_data.index, weather=fx_data)

mc.ac.plot()
plt.ylim(0, None)
plt.ylabel('AC Power (W)')
```
[image]

The spectral_modifier (spectral loss) in the bellowing code is high at 6:00 am. Multiplying spectral_modifier to get effective irradiance makes the effective irradiance high at 6:00 am, which ultimately leads to a high AC power. I am not sure if the formula to get the effective_irradiance is correct.   
```python
    def effective_irradiance_model(self):
        fd = self.system.module_parameters.get('FD', 1.)
        self.effective_irradiance = self.spectral_modifier * (
            self.total_irrad['poa_direct']*self.aoi_modifier +
            fd*self.total_irrad['poa_diffuse'])
        print('aoi_modifier')
        print(self.aoi_modifier)

        return self
```

cwhanse, 2017-08-05:
Could you post the following time series: air mass, AOI, spectral modifier.  Thanks.

[image]<https://user-images.githubusercontent.com/26880408/28986996-01d33122-792f-11e7-92a7-35b4ddf8ae38.png>

jielianguo, 2017-08-07:
Sure!

[aoi_airmass_spectralModifier.txt](https://github.com/pvlib/pvlib-python/files/1203881/aoi_airmass_spectralModifier.txt)

jielianguo, 2017-08-07:
Sure!  Thank you @cwhanse 
[aoi_airmass_spectralModifier.txt](https://github.com/pvlib/pvlib-python/files/1203883/aoi_airmass_spectralModifier.txt)

cwhanse, 2017-08-07:
The spikes in AC power result from the spectral modifier being unconstrained.  I'll have to look into how we've addressed this in Matlab and in other analyses.

jielianguo, 2017-08-07:
Got you! Thank you.

cwhanse, 2017-08-07:
The problem stems from the underlying SAPM model for the spectral modifier, which fits a polynomial (4th order) in air mass to measurements.  The pvlib implementation sets a lower bound of 0 for the modifier, but no upper bound.  For some modules, at high air mass, the function will return extremely large values for the spectral modifier.  

We don't have a replacement SAPM spectral model.  Two things you can do:
1 - switch to the FirstSolar contributed model for the spectral modifier, in pvlib/atmosphere/first_solar_spectral_correction.  I don't know how to wire that into the model chain.
2 - edit your copy of pvlib/pvsystem/sapm_spectral_loss to impose a maximum on the spectral loss modifier.  My own opinion, I think 1.1 is appropriate.

@wholmgren @jforbess  what are your thoughts on this? I don't think its a bug per se.  I think opening an issue for an enhancement to prevent sapm_spectral_loss from returning overly large values is appropriate.  Is there a way to bring the first_solar_spectral_correction forward as an option when specifying a modelchain?

wholmgren, 2017-08-07:
@jielianguo if you're using forecast data then you might not need to worry about the marginal accuracy improvements from a spectral loss model. The forecast is likely to be the limiting factor in your model accuracy. Simply set ``spectral_model='no_loss'`` and avoid all the trouble.

@cwhanse I'd support a pull request to implement a maximum value in the spectral loss function.

In the mean time, one alternative to editing the ``sapm_spectral_loss`` source code is to pass a function into the ``ModelChain`` constructor. I've not tested the code below, but it should work:

```python
def sapm_spectral_loss_clipped(mcobj):
    mcobj.spectral_modifier = mcobj.system.sapm_spectral_loss(self.airmass['airmass_absolute'])
    mcobj.spectral_modifier = np.maximum(mcobj.spectral_modifier, 1.1)

mc = ModelChain(..., spectral_model=sapm_spectral_loss_clipped)
```

I'll make another issue for bringing the ``first_solar_spectral_correction`` function up to ``ModelChain``.

adriesse, 2017-08-07:
I think the main problem is that the polynomial is not a great function form for this phenomenon, but it is often risky to use polynomial functions, especially higher orders, for extrapolation anyway.

Limiting the output certainly fixes the symptoms.  Limiting the input could be another option--one which could reflect the range of measurements that were used to obtain the coefficients.

cwhanse, 2017-08-08:
@adriesse I agree, a polynomial is a poor choice, but it was made years ago and is now baked into the parameter databases. We ought to replace it with a functional form which has constrained asymptotic behavior, but that's a discussion not for pvlib GitHub.

closed 2017-09-13 by wholmgren