pvlib/pvlib-python #359: Expose first_solar_spectral_correction in ModelChain and PVSystem

wholmgren, 2017-08-07:
I suggest the following steps to expose the ``first_solar_spectral_correction`` function to ``ModelChain``:

1. Add a ``PVSystem.first_solar_spectral_correction(pw, airmass_absolute)`` method that wraps the ``atmosphere.first_solar_spectral_correction(pw, airmass_absolute, module_type=None, coefficients=None)`` function. This is straightforward except for deciding how to handle the keyword arguments. The Sandia module database has a ``Material`` key  and the CEC database has a ``Technology`` key. These keys could be mapped to the allowed ``module_type`` arguments. I'm not sure what to do about the ``coefficients`` option. Perhaps direct users to put their custom coefficients in ``module_parameters['fs_spectral_coefficients']``?

2. Implement the stub ``ModelChain.first_solar_spectral_loss`` method. Maybe something like:
```python
    def first_solar_spectral_loss(self):
        self.spectral_modifier = self.system.first_solar_spectral_loss(
            self.weather['precipitable_water'], self.airmass['airmass_absolute'])
        return self
``` 
I'm guessing that most users don't have PW in their weather dataframes, so this line is going to be a frequent source of errors. We might consider adding a ``try/except KeyError`` with a more helpful error message.

3. Improve ``ModelChain.infer_spectral_model``. This probably amounts to setting ``mc._spectral_model = mc.first_solar_spectral_loss`` if a CEC module or a ``module_parameters['fs_spectral_coefficients']`` key is detected on the ``PVSystem`` object.

4. Optional. Add a dewpoint temperature to precipitable water conversion to the ``ModelChain.prepare_inputs`` method.

5. Add tests.

6. Update documentation.

It's unfortunate that ``sapm_spectral_loss`` and ``first_solar_spectral_correction`` live in different modules. Fixing that could be step 0, or we could leave it alone.

I'm in no rush to implement this, but I'd help anyone with a pull request.

KonstantinTr, 2017-09-08:
I'd like to takle this task. But I might need some help with taking decisions how it would be more convenient for users.

cwhanse, 2018-05-12:
@KonstantinTr are you still interested to make this pull request?

KonstantinTr, 2018-05-21:
I'm truly sorry, I can't manage to find time to deliver this :(
I've recently changed industry and coutry of residence..

cwhanse, 2018-05-21:
No worries, I didn't want this issue to stay unresolved, Many, many thanks for your contributions, and best wishes for the future.

PR #466 (2018-05-31): Expose first_solar_spectral_correction
pvlib python pull request guidelines
====================================

Thank you for your contribution to pvlib python!

You may submit a pull request with your code at any stage of completion, however, before the code can be merged the following items must be addressed:

 - [x] Closes #359
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests must pass on the TravisCI and Appveyor testing services.
 - [x] Code quality and style is sufficient. Passes ``git diff upstream/master -u -- "*.py" | flake [... cut]
closed 2018-05-31 by wholmgren