pvlib/pvlib-python #363: Overflow warning from pvlib.clearsky.haurwitz

cwhanse, 2017-08-11:
The Haurwitz model takes apparent_zenith as its argument.  pvlib ephemeris calculations can return zenith>90 for times after sunset. The Haurwitz model includes this term: 
`np.exp(-0.059/cos_zenith)`
which causes a numerical overflow when cos_zenith is small and negative, i.e., apparent_zenith is greater than but close to 90.
The current Haurwitz model code obscures the warning by 
` clearsky_ghi[clearsky_ghi < 0] = 0`
`clearsky_ghi < 0` occurs whenever `cos_zenith<0`
I propose that we apply the Haurwitz formula only for `cos_zenith>0` and set `clearsky_ghi=0` otherwise.
The function output would remain unchanged but the overflow condition would be handled.

wholmgren, 2017-08-11:
Sounds like a good approach.

cwhanse, 2017-08-12:
What would you like to do with the tests?  All the clearsky model test follow the same pattern: compute GHI at 3hr intervals for 2014/6/24 in a location in US/Arizona, compare with known result.  To catch this edge case we'd need to insert a time point just prior to sunrise or just after sunset.  I'm OK with this, but do you want to propagate the edge case to all the other clearsky models?

wholmgren, 2017-08-12:
If you want to modify the existing test function, it might be easiest to
first replace the existing times/zenith code with an explicit definition of
the zenith angles you want to test. Or you could make a simple new test
function that targets this behavior.

Consistency among all of the clear sky functions would be great, but I'll
leave it to you to decide the scope of the change.

cwhanse, 2017-08-12:
I'll rewrite the Haurwitz test to operate on a range of zenith angles.  Right now, the test could fail due to unrelated changes in the solar position algorithm.  To keep it clean I'll not touch the other clearsky model tests where this could also happen: ineichen and simplified solis.  We might want to consider more focused test cases for these other clear sky models.

wholmgren, 2017-08-15:
closed by #364

closed 2017-08-15 by wholmgren