pvlib/pvlib-python #365: QUESTION! Step-like changes in plane-of-array irradiance gradients

KonstantinTr, 2017-08-31:
I was trying to model how solar geometry and plane-of-array irradiance at noon change through the year.
I'd picked Hawaii for the modeling because they don't observe daylight saving.

Here is my code (based on [LocalizedPVSystem](https://pvlib-python.readthedocs.io/en/latest/package_overview.html#object-oriented-localizedpvsystem) section of pvlib-python.readthedocs.io documentation)

```python
import pandas as pd
import numpy as np

import pvlib
from pvlib.pvsystem import LocalizedPVSystem

temp_air = 20
wind_speed = 0

mdl = pvlib.pvsystem.retrieve_sam('SandiaMod')['Canadian_Solar_CS5P_220M___2009_']
inv = pvlib.pvsystem.retrieve_sam('cecinverter')['ABB__MICRO_0_25_I_OUTD_US_208_208V__CEC_2014_']
 
localized_system = LocalizedPVSystem(
    module_parameters=mdl,
    inverter_parameters=inv,
    surface_tilt=20,
    surface_azimuth=180,
    latitude=19.5,
    longitude=156,
    altitude=14,
    name='KOA', # airport code
    tz='US/Hawaii'
)

times = pd.DatetimeIndex(start='2015-01-01T12:00:00', end='2016-01-01T12:00:00', freq='1d')
times = times.tz_localize(localized_system.tz)

clearsky = localized_system.get_clearsky(times)

solar_position = localized_system.get_solarposition(times)

total_irrad = localized_system.get_irradiance(
    solar_position['apparent_zenith'],
    solar_position['azimuth'],
    clearsky['dni'], clearsky['ghi'], clearsky['dhi'],
    model='perez'
)
```

Then I had noticed that modeled PoA curves are not completely smooth and calculated their pseudo-gradients using [pandas.DataFrame.diff](https://pandas.pydata.org/pandas-docs/stable/generated/pandas.DataFrame.diff.html) method

[image]

Then I calculated pseudo-gradients for the solar geometry and there are no step-like changes in gradients

[image]

I've tried several other locations, module orientations and 'model' argument values having similar step-like changes.

**The question is:** Is it okay that gradual changes in solar position causes step-like changes in the plane-of-array irradiation? It seems unnatural to me.

Many thanks in advance!

P.S. I'm volunteering to contribute to the fixing process if this behavior is an issue.

adriesse, 2017-08-31:
I looks to me like the monthly turbidity values are causing this.

cwhanse, 2017-08-31:
I think Anton’s correct.  You can switch to a different clearsky model, haurwitz or simplified_solis, which doesn’t use the Linke turbidity tables.  The default in Ineichen.  It’s a keyword argument for get_clearsky(…, model=’<your choice here’>)

At one point, I thought that Will had looked into a smoothing technique for the monthly boundaries in the Linke turbidity values, but I could be wrong about that.

KonstantinTr, 2017-08-31:
@cwhanse thanks a lot for your suggestion to try a different clearsky model.

Should I close this issue?

wholmgren, 2017-08-31:
There is some limited support for smoothing out the monthly boundaries in the LT values. If I remember correctly, the LT values do not jump from month to month, but the first derivative is still discontinuous at the monthly boundaries. See clearsky.lookup_linke_turbidity. 

You can also pass your own LT values to the get_clearsky method using the ``linke_turbidity`` keyword.

KonstantinTr, 2017-09-01:
@wholmgren Many thanks for advice!

closed 2017-09-01 by KonstantinTr

jforbess, 2017-09-06:
Hm. It looks like the discontinuities occur at the mid month mark, and
therefore there is a monthly value that applies to the POA calculation,
calculated from mid month to mid month. I think this is from the Linke
Turbidity calcs:
https://pvlib-python.readthedocs.io/en/latest/generated/pvlib.clearsky.lookup_linke_turbidity.html#pvlib.clearsky.lookup_linke_turbidity

dacoex, 2017-10-09:
@KonstantinTr 
Could you add a remarkt to the docs?

@wholmgren What about a FAQ section in the docs where we point to some issues that clarify teh library?
Similar to pandas in [Cookbook](http://pandas.pydata.org/pandas-docs/stable/cookbook.html)

KonstantinTr, 2017-10-12:
@dacoex  I have no idea where to put such a remark. Somewhere close to Linke
Turbidity description?.. What was your idea?

cwhanse, 2017-10-12:
I'd add to the comment block for the Ineichen clearsky function 
https://pvlib-python.readthedocs.io/en/latest/generated/pvlib.clearsky.ineichen.html