pvlib/pvlib-python #371: New user: Memory error?

Suraj--Peri, 2017-09-06:
Hey, I just started using the library to get forecasts using a couple different methods. It was working fine until today morning and somehow gives me a memory error when I run it now. The script is below: 

`import datetime
import inspect
import os

import numpy as np
import pandas as pd

from pvlib import solarposition,irradiance,atmosphere,pvsystem
from pvlib.forecast import NDFD, RAP, HRRR
from pvlib import pvsystem

system_frame = pd.read_csv(r'C:\Users\VMItronCandA\Desktop\DER\Raw_Data_Actual\Dashboard_Database_Tracking.csv', header = 0)

demand_frame = system_frame[system_frame['Demand'] == 1]
demand_frame = demand_frame.reset_index()

if not os.path.exists(r'C:\Users\VMItronCandA\Desktop\DER\PV_Simulations\Demand_Sites\%s' % datetime.datetime.now().strftime('%Y-%m-%d_%H')):
    os.makedirs(r'C:\Users\VMItronCandA\Desktop\DER\PV_Simulations\Demand_Sites\%s' % datetime.datetime.now().strftime('%Y-%m-%d_%H'))

forecast_models = [NDFD(), RAP(), HRRR()]
fm_strings = ['NDFD', 'RAP', 'HRRR']
albedo = 0.2
for i in range(0, len(demand_frame)):
    latitude = demand_frame['latitude'][i]
    longitude = demand_frame['longitude'][i]
    tz = system_frame['tz'][i]
    today = pd.Timestamp(datetime.date.today(), tz=tz)
    start = today - pd.Timedelta(days=1)
    end = today + pd.Timedelta(days=7)
    surface_tilt = demand_frame['tilt'][i]
    surface_azimuth = demand_frame['azimuth'][i]
    for j in range(0, len(fm_strings)):
        fm = forecast_models[j]
        forecast_data = fm.get_processed_data(latitude, longitude, start, end)
        time = forecast_data.index
        a_point = fm.location
        solpos = a_point.get_solarposition(time)
        ghi = forecast_data['ghi']
        dni_extra = irradiance.extraradiation(fm.time)        
        airmass = atmosphere.relativeairmass(solpos['apparent_zenith'])
        poa_sky_diffuse = irradiance.haydavies(surface_tilt, surface_azimuth,
                                             forecast_data['dhi'], forecast_data['dni'], dni_extra,
                                             solpos['apparent_zenith'], solpos['azimuth'])
        poa_ground_diffuse = irradiance.grounddiffuse(surface_tilt, ghi, albedo=albedo)
        aoi = irradiance.aoi(surface_tilt, surface_azimuth, solpos['apparent_zenith'], solpos['azimuth'])
        poa_irrad = irradiance.globalinplane(aoi, forecast_data['dni'], poa_sky_diffuse, poa_ground_diffuse)
        temperature = forecast_data['temp_air']
        wnd_spd = forecast_data['wind_speed']
        pvtemps = pvsystem.sapm_celltemp(poa_irrad['poa_global'], wnd_spd, temperature)        
        for k in range(0, len(poa_irrad)):
            if poa_irrad['poa_global'][k] > 1000:
                poa_irrad['poa_global'][k] = poa_irrad['poa_global'][k-1]
                
        temperature = forecast_data['temp_air']
        wnd_spd = forecast_data['wind_speed']
        pvtemps = pvsystem.sapm_celltemp(poa_irrad['poa_global'], wnd_spd, temperature)
        dc_out= pvsystem.pvwatts_dc(poa_irrad['poa_global'], pvtemps['temp_cell'], system_frame['dc'][i], -.002)
        if j == 0:
            dc_frame = pd.DataFrame({'NDFD_kWh': dc_out})
        if j == 1:
            dc_frame['RAP_kWh'] =  dc_out
        if j == 2:
            dc_frame['HRRR_kWh'] = dc_out
    dc_frame['Itron_ID'] = demand_frame['ID'][i]
    dc_frame.index = dc_frame.index.tz_localize(None)
    out_name =  datetime.datetime.now().strftime('%Y-%m-%d_%H') + '\\' + demand_frame['ID'][i] + '_Sim'
    dc_frame.to_csv(r'C:\Users\VMItronCandA\Desktop\DER\PV_Simulations\Demand_Sites\%s.csv' % out_name, index_label = 'Datetime')
        `
[... 162 lines cut ...]
    532 
    533     def options(self, url, **kwargs):

C:\ProgramData\Anaconda3\lib\site-packages\requests\sessions.py in request(self, method, url, params, data, headers, cookies, files, auth, timeout, allow_redirects, proxies, hooks, stream, verify, cert, json)
    516         }
    517         send_kwargs.update(settings)
--> 518         resp = self.send(prep, **send_kwargs)
    519 
    520         return resp

C:\ProgramData\Anaconda3\lib\site-packages\requests\sessions.py in send(self, request, **kwargs)
    637 
    638         # Send the request
--> 639         r = adapter.send(request, **kwargs)
    640 
    641         # Total elapsed time of the request (approximately)

C:\ProgramData\Anaconda3\lib\site-packages\requests\adapters.py in send(self, request, stream, timeout, verify, cert, proxies)
    500                 raise ProxyError(e, request=request)
    501 
--> 502             raise ConnectionError(e, request=request)
    503 
    504         except ClosedPoolError as e:

ConnectionError: HTTPConnectionPool(host='thredds.ucar.edu', port=80): Max retries exceeded with url: /thredds/ncss/grib/NCEP/NDFD/NWS/CONUS/CONDUIT/Best/dataset.xml (Caused by NewConnectionError('<requests.packages.urllib3.connection.HTTPConnection object at 0x000000FC2489EF98>: Failed to establish a new connection: [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond',))

cwhanse, 2017-09-06:
Is there a proxy involved?  If so I’d check your proxy settings first.

in ()

Suraj--Peri, 2017-09-06:
I'm sorry, could you tell me how I'd go about checking the proxy? Thanks.

wholmgren, 2017-09-06:
@Suraj--Peri there are a number of networking errors in your traceback. It is neither a memory issue nor anything to do with pvlib.

closed 2017-09-06 by wholmgren