pvlib/pvlib-python #374: solarposition.py: args mismatch for pyephem call from get_solarposition function

KonstantinTr, 2017-09-13:
pyephem function is defined with next args:
```python
def pyephem(time, latitude, longitude, altitude=0, pressure=101325,
            temperature=12):
```
then we call it from get_solarposition.
```python
    elif method == 'pyephem':
        ephem_df = pyephem(time, latitude, longitude, pressure, temperature,
                           **kwargs)

```
as far as altitude is missing and we are passing positional args -  pressure goes to altitude,  temperature goes to pressure and temperature receives its default value

during the tests pyephem is called with keyword args and the is no test for calling get_solarposition with method='pyephem'  arg.

```python
def test_pyephem_physical(expected_solpos):
...
    ephem_data = solarposition.pyephem(times, golden_mst.latitude,
                                       golden_mst.longitude, pressure=82000,
                                       temperature=11)
...
```

I have fixed pyephem function call from get_solarposition but before creating a pull request I'd like to check what test will be sufficient for covering such a mistake? Will get_solarposition call with a method='pyephem' be enough?

wholmgren, 2017-09-13:
Good catch. I think the proposed test would be sufficient.

wholmgren, 2017-10-09:
closed by #381

closed 2017-10-09 by wholmgren