pvlib/pvlib-python #378: Tutorial resource: implement own PV module

Booli, 2017-09-26:
Hey,

what a great work this is. I am pleasantly surprised on how active this library is being maintained. I've been going through the documentations, but can't really figure out how to add my own module to simulate. 

I would like to, for example, run simulations on this very tiny module: http://ixapps.ixys.com/DataSheet/SLMD121H04L_Nov16.pdf 

How could I add these, and run a simulation on for example 4x these in series + 10 of those in parallel?

Thank you very much in advance for pointing me in the right direction.

cwhanse, 2017-09-26:
First you need to decide on a performance model for that module, and determine the appropriate parameters for that model.  The available models are the single diode model of De soto, the Sandia model and PVWatts. There are other models in common use (PVsyst, CEC, Huld) which are implemented in the Matlab PVLib but not yet in pvlib-python.  

With only the datasheet, perhaps the PVWatts model is the entry point, it requires only efficiency and a temperature coefficient for power, but can only return power.  If you are also interested in Voc and Isc, the De soto model is possible if you can get a temperature coefficient for power. The problem is getting the De soto model parameters. There's a parameter estimator in the SAM SDK but it sometimes does not return meaningful values. I could help with that part if you'd like.

If you have IV curves, there are parameter estimators for PVsyst, De soto and Huld in the Matlab PVLib.  We have a Sandia estimator which hasn't been pushed to the Matlab library.

The lack of a model fitting module is a gap with pvlib-python which I've been meaning to address, when I can make time for it.

Booli, 2017-09-26:
All help in regards to implementing this, and other similar modules, will be much appreciated. I would definitely also write a how-to document of it, which I can then share later. 

We have access to a LASS system near our facilities also, and have done several tests already on these types of modules which might be valuable input I reckon for deciding which performance model I have to choose. 

We have a basic calculation as a single diode system with Rshunt=800 and RSeries=20 in Ngspice but I feel that moving our simulation into this package(the industry standard it feels like) is more beneficial for use than using our own models. 

For now, only power is a great start. Where can I find the differences between performance models?

cwhanse, 2017-09-26:
I'd start with the API reference

http://pvlib-python.readthedocs.io/en/latest/api.html#pv-modeling

For each model there should be links to reference publications.

thunderfish24, 2017-09-27:
Hi Booli,

  It looks like the I-V curve on the (second) spec sheet is for one of the eight cells that is wired in a series of eight to form the module. Would you agree? I was able to digitize the I-V curve using [this tool](http://arohatgi.info/WebPlotDigitizer/app/), but it appears that converting the current density from the individual cells to the module is complicated by the imprecisely known areas involved.

  In a shameless plug, I provide an online "local" 5-parameter single diode model (SDM) fitting tool called [PVfit](http://pv-fit.com/sdm-l/fit). However, I am quite suspect of resistance/conductance values that derive from these fittings, however the R@oc and R@sc values from a local SDM fit seem to be more reliable. I am also suspect of many temperature coefficient measurements. That said, simple models like PVWatts can be surprisingly effective. Also note that if you are operating in an indoor spectrum, you might get bad results if assumptions are made based on measurements in outdoor spectra.

  I am currently shifting focus to double-diode models, which seem to have less model discrepancy. I would be happy to help fit I-V curve data for you, but please note that my "global" single/double diode model fittings use a model that is somewhat different than the De Soto, SAM, and PVsyst models, and I think you would still need a couple/few temperature coefficients to adapt my results to these other models.

  Again, with the caveat about indoor vs. outdoor conditions, maybe first try to validate the simplest PVWatts model against some test data taken in your expected usage conditions.

-Mark Campanelli

kanderso-nrel, 2023-02-17:
Is this closed by the addition of `ivtools.sdm.fit_desoto` and `ivtools.sdm.fit_cec_sam`?

cwhanse, 2023-02-17:
Yes this can be closed.

closed 2023-02-17 by cwhanse