pvlib/pvlib-python #386: irradiance.liujordan function doesn't use pressure arg

KonstantinTr, 2017-10-18:
I believe pressure has no usage in this function: 

[image]

The only call of liujordan is from cloud_cover_to_irradiance_liujordan of forecast.py doesn't pass pressure
[image]
 
Is there any chance that pressure should be used somewhere within function?

cwhanse, 2017-10-18:
No, atmospheric pressure is not used in the Liu – Jordan decomposition model.

If you are fixing this, please correct the spelling : tao should be tau.

Thanks!

KonstantinTr, 2017-10-18:
@cwhanse Thanks for the feedback. I'll fix tau spelling as well.

closed 2017-12-21 by KonstantinTr