pvlib/pvlib-python #387: test_clearsky.test_simplified_solis_series_elevation is declared twice

KonstantinTr, 2017-10-18:
We've got two declarations of test_clearsky.test_simplified_solis_series_elevation
[image]

During the testing we will run the one on the left:
[image]

What of these two test functions should we keep?

cwhanse, 2017-10-18:
I much prefer the shorter function on the left, because it has no dependence on other functions, e.g., get_solarposition.

Do we agree that a single elevation angle (80) is sufficient? 

I think that using a keyword argument in the call to simplified_solis, e.g., apparent_elevation=pd.Series(80), would help make this more easily understood.

wholmgren, 2017-10-18:
I'm guessing that I screwed up some copy/pasting when making these tests. 

I agree that the left is preferable. I think we want to ensure that the function returns 0s for elevation angles less than 0. A single additional value is probably good enough for that.

wholmgren, 2017-10-24:
In reviewing #388 I realized that there is [another test](https://github.com/pvlib/pvlib-python/blob/39c0862a33c3b6286ef1458a036547ed6beec399/pvlib/test/test_clearsky.py#L290-L298) that explicitly tests that the function returns 0s for elevation angles less than 0, so I retract my previous comment.

wholmgren, 2017-10-24:
closed by #388

closed 2017-10-24 by wholmgren