pvlib/pvlib-python #394: Unit tests depend on functions not being tested

cwhanse, 2017-11-02:
`test_irradiance.py` uses 
```
ephem_data = solarposition.get_solarposition(
    times, tus.latitude, tus.longitude, method='nrel_numpy')

irrad_data = tus.get_clearsky(times, model='ineichen', linke_turbidity=3)

dni_et = irradiance.extraradiation(times.dayofyear)
```
to compute known values for unit tests.  This dependence on functions not under test could cause a unit test to fail when the unit's code is unchanged.

wholmgren, 2018-08-14:
@cwhanse I reviewed the tests and I believe we can close this issue now that the PRs referenced above have been merged.

``test_atmosphere.py`` uses a similar pattern as shown in your description, but the returned values are not explicitly tested. There is already another issue, #158, for improving that test module.

cwhanse, 2018-08-14:
I agree, let's close this issue.  If we find more tests with dependencies, we can reference this conversation, and fix those tests with their own issue and PR.

closed 2018-08-14 by cwhanse