pvlib/pvlib-python #396: Add GTI DIRINT model

wholmgren, 2017-11-13:
We should consider adding an implementation of the GTI DIRINT model described in Marion 2015:

https://www.sciencedirect.com/science/article/pii/S0038092X15005757

In short, the GTI DIRINT model converts plane of array irradiance to DNI, DHI, and GHI.

I have a partial implementation that I've been sitting on for a few months. I'll try to get it into a PR soon.

jforbess, 2017-11-30:
Trying to implement this myself for a quick and dirty prototype. Any hint as to whether the iteration can be performed over the entire time series of POA or whether each time stamp has to be iterated? I am torn on how to evaluate the min D if evaluating over the entire timeseries at once.

wholmgren, 2017-11-30:
Real quick... here's my implementation: 

https://github.com/wholmgren/pvlib-python/blob/gtidirint/pvlib/irradiance.py#L1408

Off the top of my head I don't remember. I'll try to come back to this in the next few days.

cwhanse, 2017-11-30:
The iteration removes the error introduced by approximating GHI/cos(zenith) by GTI/cos(AOI).  The difference lies in the diffuse sky and ground irradiance received by a horizontal vs. tilted module. It appears that Bill applies iteration separately for each hourly value (Fig 2 in his paper), which I think is appropriate, because AOI changes with time.

I believe also that the convergence method in his paper could be improved, but I haven't got an alternate method to propose.

adriesse, 2017-11-30:
I'm sure you could do the iteration in parallel.  You would then have the same number of iterations for each time step, and your termination condition would have to be based on all time steps.  

I have no experience with this algorithm, but suspect that if you just iterate a small fixed number of times you would be adequate, or you could keep going until the biggest error for the year falls below some threshold.  Probably still way faster than looping through hour by hour.

PR #400 (2018-08-27): Add GTI DIRINT model
This PR adds an implementation of the GTI DIRINT model. 

I'm not confident about my API design/changes, but I'm making the PR now due to new interest in issue #396. See the whatsnew diff for a list of the changes to existing functions. I did this to avoid copy/pasting code, but maybe there's a better way. Let me know what you think. 

~The algorithm works well for angles less than 90 degrees. It struggles for angles greater than 90 degrees. I'm not sure if that's because I made a mistake. I've only used synthetic data to test it so far.~
Edit: it now seems to work ok for  [... cut]
closed 2018-08-27 by cwhanse