pvlib/pvlib-python #397: physicaliam returns np.array for pd.Series input

wholmgren, 2017-11-14:
Like most of our functions, physicaliam is supposed to return an output of the same type as the input (Series in, Series out). The function works with only numpy arrays internally, so it has a couple of lines to make the output into a Series, if necessary:

```python
if isinstance(aoi, pd.Series):
    iam = pd.Series(iam, index=aoi.index)
```

where ``aoi`` is supposed to be the unmodified input to the function.

PR #338 added this line:

```python
aoi = np.where(aoi == 0, zeroang, aoi)
```

but, ``np.where`` always returns an array, so the ``ifinstance`` always returns ``False``.

The long term solution is a smarter way to deal with in/out types. A short term solution is to stash a reference to the input object before reassigning the ``aoi`` object reference: 

```python
aoi_input = aoi
aoi = np.where(aoi == 0, zeroang, aoi)
# unmodified function
if isinstance(aoi_input, pd.Series):
    iam = pd.Series(iam, index= aoi_input.index)
```

adriesse, 2017-11-14:
I'd be tempted to set a flag at the top.

cwhanse, 2017-11-14:
Let's make a copy of `aoi `with type array, and preserve `aoi `as it is input.

cwhanse, 2017-11-14:
Now that I'm reading the signature I really want to edit the description.

wholmgren, 2017-11-14:
@cwhanse what do you want to edit about the description?

cwhanse, 2017-11-14:
Punctuation on lines 755-756, lines 758-760 could be more clear, and the paragraph at 796-798 is misplaced and confusing.  Theta is not a term used in this code; its a symbol in the reference paper.  I think this is leftover from Matlab.

wholmgren, 2018-01-23:
This was closed long ago by #398.

closed 2018-01-23 by wholmgren