pvlib/pvlib-python #402: Historical GFS data.

asanchezc91, 2018-01-04:
Hi!

I've just start with this library, and it looks great! 
I would like to obtain historical data from GFS to train my own models, but if I try to get data from 2016 I get the next error: KeyError: 'content-type'

 I've also tryed to get the data directly from noaa in order to process it. The problem is that I don't know which variables should I take to rename as raw_data.

Thank you so much!

wholmgren, 2018-01-04:
The forecast.py module is configured to access recent forecasts from a Unidata server. Last I looked, the Unidata server only archived the most recent 2-4 weeks of GFS data. You probably need to use NOMADS to get older data. I wrote a script for doing that:

https://github.com/wholmgren/get_nomads

It's far from perfect and I don't provide support for it, but you're welcome to use it.

asanchezc91, 2018-01-08:
Thanks for the script! This library it's really usefull for me. Great job!

closed 2018-01-23 by wholmgren