pvlib/pvlib-python #403: KeyError: 'linke_turbidity'

mattfriend, 2018-01-04:
Hi there. I'm a new programmer - so new to pvlib, python, and computer science generally, so apologies if I'm wasting your time with this question - and please go easy with me!

As part of a computer science course I've written a little python and javascript app that plots all of the PV site >1MW in the UK on a map (taken from a government list of all the >1MW sites in the UK), and when you click on any site, the app looks up the latitude and longitude for that specific site and returns forecast power generation for the site, using pvlib's ModelChain() and get_processed_data() methods and the GFS model. (I forecast power generation for a "canonical" installation since I don't actually know the specific details for each site. I realise this doesn't use much of the sophistication of the library, but it's sufficient for my purposes).

This was all working fine until Dec 4th 2017, at which point the code stopped working and I now get the following errors:
[image]
[image]
[image]
[image]

Tbh I don't really understand the error messages! I've figured out that there may be a problem with:
(a) the app's ability to reach certain urls - at least I think that's what the exceptions and resource warnings are! My hypothesis is that these are related to the pvlib library, though I'm not sure; 
and/or
(b) the app's ability to access either pvlib, or scipy (or both). On this, I've gone back and re-installed both libraries.

The only thing that I know of that has changed in my environment is that my IDE - the web-based Cloud9 IDE - was taken over by Amazon Web Services on or around the date the problems started happening.

I'd really appreciate any help you can give me to solve the problem! Many thanks in advance.

wholmgren, 2018-01-04:
The problem is likely related to pvlib's ability to use scipy to lookup the linke turbidity values on your platform. I think the problem in the scipy/platform part of that stack, not the pvlib part. You can get around it by specifying the linke_turbidity in your get_clearsky calls.

mattfriend, 2018-01-06:
Hi Will. Thanks for responding to me so quickly - I really appreciate it.
I will see if I can solve the problem in the scipy/platform part of my stack; and in addition I'll look into specifying the linke_turbidity in the get_clearsky calls as you suggest. I've no idea how to do this, but I know the documentation is very detailed so I'm sure I'll find a way to work it out!
Many thanks again for your help.

closed 2018-01-23 by wholmgren

JoshuaC3, 2019-01-08:
I am getting this problem also. It seems the issue is with numpy here

```
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
~/miniconda3/envs/satellite-solar-forecast/lib/python3.7/site-packages/pvlib/location.py in get_clearsky(self, times, model, solar_position, dni_extra, **kwargs)
    205             try:
--> 206                 linke_turbidity = kwargs.pop('linke_turbidity')
    207             except KeyError:

KeyError: 'linke_turbidity'

During handling of the above exception, another exception occurred:

ValueError                                Traceback (most recent call last)
<ipython-input-28-4c4d083c84a4> in <module>()
      3 train_dates = train_ghi.index
      4 
----> 5 clearsky_ghi = location.get_clearsky(train_dates)

~/miniconda3/envs/satellite-solar-forecast/lib/python3.7/site-packages/pvlib/location.py in get_clearsky(self, times, model, solar_position, dni_extra, **kwargs)
    209                 linke_turbidity = clearsky.lookup_linke_turbidity(
    210                     times, self.latitude, self.longitude,
--> 211                     interp_turbidity=interp_turbidity)
    212 
    213             try:

~/miniconda3/envs/satellite-solar-forecast/lib/python3.7/site-packages/pvlib/clearsky.py in lookup_linke_turbidity(time, latitude, longitude, filepath, interp_turbidity)
    214 
    215     if interp_turbidity:
--> 216         linke_turbidity = _interpolate_turbidity(lts, time)
    217     else:
    218         months = time.month - 1

~/miniconda3/envs/satellite-solar-forecast/lib/python3.7/site-packages/pvlib/clearsky.py in _interpolate_turbidity(lts, time)
    275     # Then we map the month value to the day of year value.
    276     # Do it for both leap and non-leap years.
--> 277     lt_leap = np.interp(dayofyear, days_leap, lts_concat)
    278     lt_no_leap = np.interp(dayofyear, days_no_leap, lts_concat)
    279     linke_turbidity = np.where(isleap, lt_leap, lt_no_leap)

~/miniconda3/envs/satellite-solar-forecast/lib/python3.7/site-packages/numpy/lib/function_base.py in interp(x, xp, fp, left, right, period)
   1306         fp = np.concatenate((fp[-1:], fp, fp[0:1]))
   1307 
-> 1308     return interp_func(x, xp, fp, left, right)
   1309 
   1310 

ValueError: object too deep for desired array
```

Seems like there should be a pvlib fix for this, especially if `get_clearsky()` is intended as a convenience.function

wholmgren, 2019-01-08:
Can you provide a minimum reproducible example?

JoshuaC3, 2019-01-08:
No - I've tried and tried but I cannot reproduce this without passing you the csv I am using!

I have looked at this through pdb and believe this is the problem:

```
ipdb> fp
array([[[55, 66, 62, 75, 79, 76, 77, 71, 74, 68, 62, 50]],

       [[55, 66, 62, 75, 79, 76, 77, 71, 74, 68, 62, 50]],

       [[55, 66, 62, 75, 79, 76, 77, 71, 74, 68, 62, 50]]], dtype=uint8)

ipdb> fp.shape
(3, 1, 12)
```

I believe, `dayofyear`, `days_leap` and `lts_concat` are being concatenated incorrectly. They should be creating a 2D array not a 3D array as above.

Does this seem correct/help? :)

cwhanse, 2019-01-08:
from `clearsky._interpolate_turbidity` could you post the argument `lts` that is passed in, and the output of line 262  `lts_concat = np.concatenate([[lts[-1]], lts, [lts[0]]])`?

JoshuaC3, 2019-01-09:
@cwhanse lts was a 2D array. I have created a fix to reshape lts. I have done PR for this, with test for both 1D and 2D cases.

JoshuaC3, 2019-01-10:
@cwhanse Here is the datetimeindex that is causing the issues. I couldn't find what was wrong with it upstream.

<details>
2016-10-15 15:00:00
2016-10-15 15:15:00
2016-10-15 15:30:00
2016-10-15 15:45:00
2016-10-15 16:00:00
2016-10-15 16:15:00
2016-10-15 16:30:00
2016-10-15 16:45:00
2016-10-15 17:00:00
2016-10-16 06:15:00
2016-10-16 06:30:00
2016-10-16 06:45:00
2016-10-16 07:00:00
2016-10-16 07:15:00
2016-10-16 07:30:00
2016-10-16 12:45:00
2016-10-16 13:00:00
2016-10-16 13:15:00
2016-10-16 13:30:00
2016-10-16 13:45:00
2016-10-16 14:00:00
2016-10-16 14:15:00
2016-10-16 14:30:00
2016-10-16 14:45:00
2016-10-16 15:00:00
2016-10-16 15:15:00
2016-10-16 15:30:00
2016-10-16 15:45:00
2016-10-16 16:00:00
2016-10-16 16:15:00
2016-10-16 16:30:00
2016-10-16 16:45:00
2016-10-17 06:30:00
2016-10-17 06:45:00
2016-10-17 07:00:00
2016-10-17 07:15:00
2016-10-17 07:30:00
2016-10-17 07:45:00
2016-10-17 08:00:00
2016-10-17 08:15:00
2016-10-17 08:30:00
2016-10-17 08:45:00
2016-10-17 09:00:00
2016-10-17 09:15:00
2016-10-17 09:30:00
2016-10-17 09:45:00
2016-10-17 10:00:00
2016-10-17 10:15:00
2016-10-17 10:30:00
2016-10-17 10:45:00
2016-10-17 11:00:00
2016-10-17 11:15:00
2016-10-17 11:30:00
2016-10-17 11:45:00
2016-10-17 12:00:00
2016-10-17 12:15:00
2016-10-17 12:30:00
2016-10-17 12:45:00
2016-10-17 13:00:00
2016-10-17 13:15:00
2016-10-17 13:30:00
2016-10-17 13:45:00
2016-10-17 14:00:00
2016-11-15 09:15:00
2016-11-15 09:30:00
2016-11-15 09:45:00
2016-11-15 10:00:00
[... 9097 lines cut ...]
2017-07-31 13:45:00
2017-07-31 14:00:00
2017-07-31 14:15:00
2017-07-31 14:30:00
2017-07-31 14:45:00
2017-07-31 15:00:00
2017-07-31 15:15:00
2017-07-31 15:30:00
2017-07-31 15:45:00
2017-07-31 16:00:00
2017-07-31 16:15:00
2017-07-31 16:30:00
2017-07-31 16:45:00
2017-07-31 17:00:00
2017-07-31 17:15:00
2017-07-31 17:30:00
2017-07-31 17:45:00
2017-07-31 18:00:00
2017-07-31 18:15:00
2017-07-31 18:30:00
2017-07-31 18:45:00
2017-07-31 19:00:00
</details>

I agree, this is absolutely treating the symptom but this IMO isn't a bad thing. If you do not like the unconditional reshaping, I could stick it in a try except statement to acknowledge the fact this is undesired behaviour?

cwhanse, 2019-01-10:
It's clear we're using different numpy versions - what version you on?  Unless we can reproduce the behavior and understand why _interpolate_turbidity is not returning a 1d object corresponding to time, I am not willing to merge the changes.

JoshuaC3, 2019-01-10:
my numpy version is 1.15.1

cwhanse, 2019-01-10:
As do I, with either numpy 1.14 or 1.15.  I think the issue is upstream in the workflow before the call to `location.get_clearsky`.  Is time a pandas.DatetimeIndex (a numpy array perhaps)?  You posted the values above, but not the data type.  Perhaps you are passing an array of lat, lon rather than a single pair?  At this point, I don't see a way to help diagnose the issue without seeing your code.

wholmgren, 2019-01-11:
A few observations:

1. Possibly a pandas version issue
2. DST issue? possibly leading to repeated values? what is the timezone?
2. Can you reproduce the error if you adapt your code to use the lower level functions directly?
2. We really do need a minimum reproducible example
2. I don't want to try to treat the symptom without at least understanding the cause
2. The issue raised by @JoshuaC3 is unrelated to the original issue except for the top-level exception. Perhaps we should add a more informative error message.

wholmgren, 2019-01-11:
> location = pvlib.location.Location(np.array([42.1]), np.array([2.2])) # breaks

This behavior is not supported. I'd like to understand your use case to see if something needs to change or if we can recommend and alternative.

JoshuaC3, 2019-01-11:
@wholmgren - it's not a use case. It's a mistake. I am happy to use the former, however, the error messages could be better for this.

I would also question why using the two arrays returns a nx1 array in the first place?

cwhanse, 2019-01-11:
I'm glad we've tracked down the cause.  We could add some type checking and conversion to `pvlib.Location.__init__` for latitude and longitude.  

I don't know why having 1D np.arrays instead of float results in that extra dimension from np.interp

mikofski, 2019-01-11:
IMO please don't use type checking. The Pythonic way is it's [easier to ask forgiveness than permission](https://docs.python.org/3/glossary.html#term-eafp) rather than type checking which slows down code and makes it less readable with a lot of branching conditions that may or may not even capture all edge cases.

IMO if you want to return a better error message use a `try: except` around the call and raise an exception with a more meaningful message

mikofski, 2019-01-11:
Also the documentation serves as a contract with the user. IMO less code is better. If only scalar Python natives can be used, then the docstring should say that, and the user should refer there

mikofski, 2019-01-11:
We can add a troubleshooting section to the docs

wholmgren, 2019-01-11:
I agree with @mikofski. The [Location](https://pvlib-python.readthedocs.io/en/latest/generated/pvlib.location.Location.html#pvlib.location.Location) documentation says that these parameters must be `float`. Perhaps it's reasonable to expect that other scalar types should work but, in my opinion, it's not reasonable to expect that length-1 arrays should work.

Improving this error handling may actually require removing a try/except error. The `KeyError` may be misleading users.

```python
# existing code
            try:
                linke_turbidity = kwargs.pop('linke_turbidity')
            except KeyError:
                interp_turbidity = kwargs.pop('interp_turbidity', True)
                linke_turbidity = clearsky.lookup_linke_turbidity(
                    times, self.latitude, self.longitude,
                    interp_turbidity=interp_turbidity)

# alternative code
            linke_turbidity = kwargs.pop('linke_turbidity', None)
            if linke_turbidity is None:
                interp_turbidity = kwargs.pop('interp_turbidity', True)
                try:
                    linke_turbidity = clearsky.lookup_linke_turbidity(
                        times, self.latitude, self.longitude,
                        interp_turbidity=interp_turbidity)
                except Exception as e:
                    raise e('Could not lookup linke turbidity. Check input types or call get_clearsky with valid  linke_turbidity keyword argument')
```

cwhanse, 2019-01-11:
> IMO if you want to return a better error message use a try: except around the call and raise an exception with a more meaningful message.

Agree. In this case, the user is a victim of duck-typing (passed a singleton numpy.array where a float was expected).

cwhanse, 2019-01-11:
I'd go deeper, into `lookup_linke_turbidity`

`    latitude_index = (
        np.around(_linearly_scale(latitude, 90, -90, 0, 2160))
        .astype(np.int64))
    longitude_index = (
        np.around(_linearly_scale(longitude, -180, 180, 0, 4320))
        .astype(np.int64))`

When `latitude` is float, `latitude_index` is float and the indexing into the h5 file returns a 1d numpy array.  When `latitude` is a 1d array (even a singleton), `latitude_index` is returned as a 3d numpy array. There's no exception raised in either case, but here's where we could detect the problem.

cwhanse, 2019-01-11:
Or a try/except around 

linke_turbidity = _interpolate_turbidity(lts, time)