pvlib/pvlib-python #404: build, test, and deploy and deploy for mac os

mikofski, 2018-01-04:
The most current version of pvlib for Mac OS on anaconda is v0.4.5, even though the most current version for Windows and Linux is v0.5.1 (see https://anaconda.org/pvlib/pvlib) - I assume this is because no one who has sufficient priviledges on Anaconda has a Mac OS.

However, Travis-CI will built, test, and deploy on both Linux and Anaconda. Therefore adding these lines:

```yml
os:
  - linux
  - osx
```

will test both according to [Travis-CI docs](https://docs.travis-ci.com/user/multi-os/) - should we try it?

wholmgren, 2018-01-04:
Yikes. I build all of the conda packages on my mac, and I'm not sure why they haven't been uploaded for the last few versions. I guess I didn't notice because I only ever install from source on my mac. I'll try to get those uploaded this afternoon or tomorrow.

As far as testing goes, I agree that it would be good to add the osx tests.

On a related note, I had some issues with the builds for the conda-forge channel awhile ago and gave up on it for now. I'd be happy to give you write access to that repository if you're interested in picking it up.

I'm less sure about the "deploy" part of your comment for our own CI configs. The conda forge experience has left me less than enthusiastic about automated deployment tools. Obviously I've screwed up the manual deployment, too, so nothing is perfect. Do your tools use Travis to deploy?

wholmgren, 2018-01-04:
I uploaded the 0.5.0 and 0.5.1 osx packages to the pvlib channel and confirmed that they work on my mac.

mikofski, 2018-01-10:
Thanks for uploading the Mac os builds!

Sure I can try to look at the conda forge builds, altho I am just learning about how to make feedstocks.

Yes, most of my tools are deployed by Travis CI to both GitHub releases and pypi automatically whenever I push a tag to GitHub. I have something like versioneer to automatically update my `setup.py`

mikofski, 2018-06-20:
[Travis-ci doesn't support Python runtime on osx as of April 7, 2018](https://github.com/travis-ci/travis-ci/issues/2312)

closed 2018-06-20 by mikofski

mikofski, 2018-06-20:
@wholmgren did you want to keep this open, just notices the **someday** tag. up to you.