pvlib/pvlib-python #411: Test singlediode() implementations using a "gold" function and data set

thunderfish24, 2018-01-29:
#409 Suggests Bishop [1988] as a truly explicit method to calculate i-from-v. Bishop's method is not subject to Newton convergence failure or to LambertW argument "blow up", handles ideal device parameters without extra logic, and has no dependency on scipy, and naturally vectorizes via standard numpy functions' vectoization. The trade-off is that the explicit device terminal current computation occurs at the diode junction voltage (V_d := V + I*R_s), not the device terminal voltage. (The device terminal voltage is "back-calculated" as V = V_d - I*Rs.) Thus, additional computation work must be done to determine currents at specified terminal voltages, inc. I_sc @ V_sc=0, I_mp @ V_mp, and I_oc=0 @ V_oc.

The Bishop method is a very good candidate for a "gold" function implementation for testing the accuracy and speed of [various implementations of `i_from_v()`](#410) that pvlib may wish to offer users or refactor to. By inverting the dataset, it could also be used to test `v_from_i()` implementations.
```python
singlediode(photocurrent, saturation_current, resistance_series, resistance_shunt, nNsVth, ivcurve_pnts=None)
```

I think that after this `bishop88` function is implemented (presumably by #409), we should create and commit a corresponding "gold" dataset that we can use for better testing, refactoring, and optimization of the `singlediode()` function. I think we should try to cover a large number (all?) of the devices in the CEC database over a combination of temperature and irradiance ranges (perhaps the IEC matrix?). This dataset should also be valuable to the greater PV modeling community who wish to test/contribute new computational algorithms.

Outstanding questions:
(1) What should the voltage range be? Specifically, the diode voltages will give some terminal voltages less than zero, and how much beyond V_oc should we try to extend?
(2) Can/should we also include I_sc, I_x, I_mp, I_xx, and V_oc points, and what about the slope at I_sc and slope at V_oc?
(3) Does python have an interval analysis package up to the task of verifying our computations, e.g., [intPy](https://pypi.python.org/pypi/IntPy)?
(4) Is it feasible to run such a comprehensive test suite on every PR?

cc @mikofski @adriesse

wholmgren, 2018-01-30:
I am a big +1 on testing pvlib against "gold datasets" with every PR. The implementation probably depends on the size and scope of a dataset. Adding numeric checks to the test suite to guard against unknown changes was a big effort, so I'd recommend starting small.

adriesse, 2018-01-31:
I don't think ``singlediode()`` is the right hierarchy level for verifying algorithms and implementations.  I would prefer to see functions like ``Voc()``, ``Isc()``, and especially ``MPP()`` in addition to ``I_from_V()`` and ``V_from_I()``.  For all of these there can be reference implementations and others.  ``singlediode()`` just seems like a convenient way to call several of these in sequence, while most of the time they are non needed.

cwhanse, 2018-01-31:
I agree with Anton on the factorization.

thunderfish24, 2018-02-01:
I’m getting started on this. I’m expecting a few PRs out of it as I start out small.

thunderfish24, 2018-02-03:
I have decided to start with creating minimal "gold" I-V dataset for perhaps a half dozen or so model parameter sets.

Bear with me as this is my first real foray into reliable computing. Using [pyinterval](http://pyinterval.readthedocs.io/en/latest/guide.html), below is a notebook snippet where I generate I-V data pairs using the bishop88 technique for a SunPower module, then I calculate the residuals using enclosing intervals for all the computed pairs and model parameters. This produces list of residual intervals, all of which contain zero and which have a width that gives one more/less confidence as to the validity of the solution pair. So, I think asserting zero in the interval and the interval endpoints not too far from zero could be the acceptance criterion for creating the gold dataset. Using the dataset would then entail running a particular choice of i_from_v() or v_from_i() and verifying that the answer is not too far from the gold values (no interval analysis used here, keeping this fairly straightforward). I am still wondering about the spacing and breadth of the I-V pairs.

Feedback here is welcomed.

```python
v_oc_V = pvsystem.v_oc(rsh, rs, nnsvt, i_rs_A, i_ph_A)
v_d_V = np.linspace(0., v_oc_V, 11)
i_A = i_ph_A - i_rs_A*np.expm1(v_d_V/nnsvt) - g_p_S*v_d_V
v_V = v_d_V - r_s_Ohm*i_A
for sol_pair in zip(v_V, i_A):
    print("{:+20.16f} V, {:+20.16f} A".format(sol_pair[0], sol_pair[1]))
>>>
 -2.1696985296000002 V,  +5.8640500800000002 A
 +3.6145797904049686 V,  +5.8468400769024944 A
 +9.3988583628994498 V,  +5.8296293914008999 A
+15.1831387169629082 V,  +5.8124138908480232 A
+20.9674316417980293 V,  +5.7951644152365844 A
+26.7518132661460761 V,  +5.7776752112118412 A
+32.5368207553018678 V,  +5.7584944806796940 A
+38.3262443539114628 V,  +5.7273783191913070 A
+44.1468280728327542 V,  +5.6120456163199428 A
+50.1872779676257110 V,  +4.9024800056873206 A
+57.7791061885889121 V,  +0.0000000000000025 A

interval_res_A_list = [ interval(i_ph_A) - interval(i_rs_A)*\
    imath.expm1((interval(sol_pair[0]) + interval(sol_pair[1])*interval(rs))/interval(nnsvt)) - \
    interval(g_p_S)*(interval(sol_pair[0]) + interval(sol_pair[1])*interval(rs)) - interval(sol_pair[1]) \
    for sol_pair in zip(v_V, i_A) ]
interval_res_A_list
>>>
[interval([-0.0, 1.7763568394002505e-15]),
 interval([-8.881784197001252e-16, 8.881784197001252e-16]),
 interval([-0.0, 1.7763568394002505e-15]),
 interval([-1.7763568394002505e-15, 0.0]),
 interval([-8.881784197001252e-16, 8.881784197001252e-16]),
 interval([-8.881784197001252e-16, 8.881784197001252e-16]),
 interval([-0.0, 1.7763568394002505e-15]),
 interval([-8.881784197001252e-16, 8.881784197001252e-16]),
 interval([-8.881784197001252e-16, 1.7763568394002505e-15]),
 interval([-8.881784197001252e-16, 6.217248937900877e-15]),
 interval([-2.223221606811876e-14, 2.1316282072803006e-14])]
```

mikofski, 2018-02-03:
Couldn't you also use the [numerical precision test](../../../mikofski/pvlib-python/blob/faster_way/pvlib/test/test_numerical_precision.py#L40) to generate exact values?

thunderfish24, 2018-02-03:
Might be nothing more than a matter of approach, but my numerical background is more in terms of thinking in terms of intervals with exactly represented floating point endpoints containing a true value, rather than arbitrary precision “objects”.

thunderfish24, 2018-02-04:
Looks like the built in (decimal package)[https://docs.python.org/3/library/decimal.html#mitigating-round-off-error-with-increased-precision] is another alternative.

mikofski, 2018-02-04:
No, IMO the decimal type is not good for science and engineering.

mikofski, 2018-02-04:
Ah, my mistake, you mean for creating the gold data set, yes, decimal will be similar to sympy symbols, except that you need to specify the precision of decimal manually, whereas sympy will adjust that automatically because it auto differentiates

thunderfish24, 2018-02-05:
Here's a gist of my beta I-V curve dataset creator: https://gist.github.com/thunderfish24/10bd9e57e0cb1414fc79601f53f16178

Note the tolerance interval on the residuals that I was able to achieve for the two modules considered (with ideal and degraded versions added) is [-1.e-12, 1.e-12]. I-V curves are engineered to fall "slightly" outside the first quadrant. (I have some unexpected import warnings to track down too.)

I really want to add a x-Si cell in the list of devices, then I will move on to saving the dataset in a lossless interchangeable format, as well as writing a generic function tester (possibly with timing info).

I think a second PR will add the "derived" parameters, such as Isc, Ix, R@Isc, Pmp, Ixx, R@Voc, Voc, and FF, but I will try to make the dataset format extensible so that adding things doesn't break existing consumers.

mikofski, 2018-02-05:
@thunderfish24, this it's really great!.Thanks so much for your hard work.

thunderfish24, 2018-02-08:
I have some first results comparing the existing i_from_v() and v_from_i() computations against the verified gold values computed using Bishop's explicit method. *Preliminary* results suggest that v precision can be surprising low for some parameter combo's. More details and code for others to run to follow.

mikofski, 2018-02-08:
Voltage precision will be very difficult where `di/dv` is very large, _eg_ approaching `v_oc` that's just a fact of life and why we don't bother trying hard in quadrant 4. Reverse bias is easier because at least you know it approaches an asymptote.

cwhanse, 2018-02-08:
The v_from_i calculation is where the overflow issues occur in the lambertw scheme.  There could certainly be a precision loss there.  I also suspect that we can't expect the circle of (I, V from Bishop) then (recover V from I) to maintain precision.  Suppose we know Vd precisely.  The corresponding current will not be known as precisely because di/dv is very large as @mikofski points out.  Precision in V will further declines when that current value I is fed back into v_from_i.  I guess my point is that the v_from_i method has a precision, which I inferred from the precision of the lambertw calculation (verified to within 8 decimal places) but not the end-to-end algorithm.

adriesse, 2018-02-11:
Perhaps we should define the allowable range for each of the parameters of the gold implementation.  I think such ranges should extend beyond what the module database contains for commercial modules, for example, the resistances could both be allowed any non-negative value--perhaps even some negative ones. (I have a vague recollection than some fits produce negative resistances.)

thunderfish24, 2018-02-12:
@adriesse I have seen my method try to fit negative series resistances for the SDM if I do not constrain to positive values. esp. in certain parts of the irradiance-temperature space. I am trying to see if a comparable double-diode model (DDM) suffers the same deficiency. This is why I think it can be tricky trying to implement a more physically realistic auxiliary equation for series resistance in the SDM, because the model itself may be so wrong that it confounds such efforts.

mikofski, 2018-02-12:
I agree with both @adriesse and @thunderfish24 , and it's good practice. So far I haven't transformed any of the variable going into the Newton or bisection methods. Typically I use `ustar = u**2` because that will always be positive, then `u = sqrt(ustar)` it's also positive

thunderfish24, 2018-02-12:
K. I squashed some early bugs, and now I am 95% confident in this early result for checking i_from_v and v_from_i for a small area (2 cm x 2 cm) Si cell based on reference parameters in Bishop' paper.

'v_test_current_sum_max_abs_res': 0.009735069744962271 is the maximum absolute residual of the sum of currents at the diode node for a curve where 'G': 200.0, 'T': 50.0, over 25 points.

'v_test_max_abs_res': 0.27325988859491335 is the maximum absolute residual of the terminal voltage for the same curve where 'G': 200.0, 'T': 50.0 over 25 points. This is pretty big relative to the Voc ~0.46.

I will be diving into this anomaly soon, but I welcome independent verification using the parameters logged below. I really do hope to push some code up in a PR soon too. I *think* I'm close to a reasonable "starter" API for this that can be extended and implemented in unit tests.

```bash
Bishop_ref_cell
[{'name': 'Bishop_ref_cell', 'Technology': 'Mono-c-Si', 'N_s': 1, 'i_from_v': {'current_sum': {'G': 1000.0, 'T': 75.0, 'i_l': 0.34609999999999996, 'i_0': 3.0407536094957474e-05, 'r_s': 26.6, 'r_sh': inf, 'nNsVth': 0.04500185335353259, 'v_gold': array([-1.508944  , -1.10326999, -0.79608261, -0.54685273, -0.3361162 ,
       -0.15294237,  0.00945154,  0.15557839,  0.28860004,  0.41082186,
        0.52397906,  0.62941192,  0.72817826,  0.8211283 ,  0.90895612,
        0.99223608,  1.07144908,  1.147002  ,  1.2192423 ,  1.28846916,
        1.35494214]), 'i_gold': array([ 0.07213305,  0.05697325,  0.04549074,  0.03617283,  0.02829288,
        0.02144272,  0.01536904,  0.00990328,  0.00492732,  0.00035503,
       -0.00387841, -0.00782309, -0.01151854, -0.01499653, -0.018283  ,
       -0.02139942, -0.02436376, -0.02719123, -0.02989483, -0.03248572,
       -0.03497363]), 'i_test': array([ 0.07213305,  0.05697325,  0.04549074,  0.03617283,  0.02829288,
        0.02144272,  0.01536904,  0.00990328,  0.00492732,  0.00035503,
       -0.00387841, -0.00782309, -0.01151854, -0.01499653, -0.018283  ,
       -0.02139942, -0.02436376, -0.02719123, -0.02989483, -0.03248572,
       -0.03497363]), 'i_test_current_sum_res': array([ 8.82627305e-15,  0.00000000e+00,  0.00000000e+00,  9.99200722e-15,
        0.00000000e+00,  2.08721929e-14,  1.06581410e-14,  1.08801856e-14,
        1.09356968e-14,  2.28150832e-14,  2.31481501e-14,  1.13797860e-14,
        0.00000000e+00,  1.22124533e-14,  1.23789867e-14,  0.00000000e+00,
        5.55111512e-16,  1.19904087e-14, -7.21644966e-16, -7.21644966e-16,
        1.36002321e-14]), 'i_test_current_sum_max_abs_res': 2.3148150063434514e-14}, 'terminal': {'G': 1100.0, 'T': 75.0, 'i_l': 0.38071, 'i_0': 3.0407536094957474e-05, 'r_s': 5.32, 'r_sh': inf, 'nNsVth': 0.04500185335353259, 'v_gold': array([-0.28623497, -0.16501395, -0.07096389,  0.00675164,  0.07343886,
        0.13212299,  0.18470473,  0.23246219,  0.27629915,  0.31688004,
        0.35470866,  0.39017676,  0.42359546,  0.45521625,  0.48524551,
        0.51385489,  0.54118875,  0.5673698 ,  0.59250324,  0.61668001,
        0.6399793 ]), 'i_gold': array([ 0.13007972,  0.10800769,  0.09084533,  0.07664177,  0.06443937,
        0.05369119,  0.04405308,  0.03529338,  0.0272481 ,  0.01979657,
        0.01284725,  0.00632887,  0.00018485, -0.00563062, -0.01115513,
       -0.01641994, -0.0214514 , -0.02627187, -0.03090054, -0.03535402,
       -0.03964677]), 'i_test': array([ 0.13007972,  0.10800769,  0.09084533,  0.07664177,  0.06443937,
        0.05369119,  0.04405308,  0.03529338,  0.0272481 ,  0.01979657,
        0.01284725,  0.00632887,  0.00018485, -0.00563062, -0.01115513,
       -0.01641994, -0.0214514 , -0.02627187, -0.03090054, -0.03535402,
       -0.03964677]), 'i_test_abs_res': array([5.55111512e-17, 0.00000000e+00, 0.00000000e+00, 0.00000000e+00,
       5.55111512e-17, 5.55111512e-17, 0.00000000e+00, 5.55111512e-17,
       1.11022302e-16, 0.00000000e+00, 0.00000000e+00, 5.55111512e-17,
       0.00000000e+00, 5.55111512e-17, 0.00000000e+00, 0.00000000e+00,
       0.00000000e+00, 5.55111512e-17, 0.00000000e+00, 5.55111512e-17,
       5.55111512e-17]), 'i_test_max_abs_res': 1.1102230246251565e-16}}, 'v_from_i': {'current_sum': {'G': 200.0, 'T': 50.0, 'i_l': 0.04661, 'i_0': 1.072213666670876e-06, 'r_s': 5.32, 'r_sh': 3000.0, 'nNsVth': 0.04177035447707613, 'v_gold': array([-0.00898318,  0.02848714,  0.05723074,  0.08156525,  0.1035108 ,
        0.12418598,  0.14428116,  0.16425641,  0.18443643,  0.20506097,
        0.22631354,  0.2483389 ,  0.27125414,  0.29515597,  0.32012581,
        0.34623323,  0.3735386 ,  0.40209485,  0.43194897,  0.46314305,
        0.4957151 ]), 'i_gold': array([ 0.04622058,  0.04579657,  0.04518026,  0.04435788,  0.04331858,
        0.04205327,  0.04055404,  0.03881386,  0.03682638,  0.03458579,
        0.03208671,  0.02932415,  0.02629343,  0.02299015,  0.01941012,
        0.01554941,  0.01140424,  0.00697101,  0.00224628, -0.00277325,
       -0.00809077]), 'v_test': array([-0.00898318,  0.02848714,  0.05723074,  0.08156525,  0.1035108 ,
        0.12418598,  0.14428116,  0.16425641, -0.08882346,  0.20506097,
        0.22631354,  0.2483389 ,  0.27125414,  0.29515597,  0.32012581,
        0.34623323,  0.3735386 ,  0.40209485,  0.43194897,  0.46314305,
        0.4957151 ]), 'v_test_current_sum_res': array([-4.16333634e-17, -9.02056208e-17, -1.73472348e-16, -2.77555756e-16,
       -3.19189120e-16, -4.57966998e-16, -8.32667268e-16, -1.17961196e-15,
[... cut ...]
        0.34623323,  0.3735386 ,  0.40209485,  0.43194897,  0.46314305,
        0.4957151 ]), 'v_test_abs_res': array([4.38538095e-15, 4.60742555e-15, 5.38458167e-15, 5.41233725e-15,
       4.21884749e-15, 4.32986980e-15, 5.85642645e-15, 6.38378239e-15,
       2.73259889e-01, 3.38618023e-15, 4.16333634e-15, 5.85642645e-15,
       1.07691633e-14, 9.38138456e-15, 1.83186799e-14, 1.04916076e-14,
       1.99840144e-15, 8.88178420e-16, 8.71525074e-15, 3.21964677e-15,
       1.76525461e-14]), 'v_test_max_abs_res': 0.27325988859491335}}}]
```

cwhanse, 2018-02-12:
@thunderfish24 that's really odd behavior, at one voltage value the residual is 13 orders of magnitude greater than at any other value.  Any chance you can fix the text wrapping in the posted code?

thunderfish24, 2018-02-12:
Sorry I posted that hastily on my lunch break. I will try to fix soon.

thunderfish24, 2018-02-12:
I updated the formatting. To be clear, these are the four I-V curves out of the (G,T) matrix that produced the worst maximum absolute residual across the I-V values in each curve for the specific function (`i_from_v()` or `v_from_i()`), for both the residual for the current sum at the diode node (ideally zero) and for the calculated terminal value's difference from the gold value.

thunderfish24, 2018-02-13:
I updated the json output above to have signed residuals that are consistent between the various computations.

cwhanse, 2018-02-13:
@thunderfish24 I think there's something amiss in your calculation of v_test.  In the 8th position you get a value of -0.0888234583382328.  Using the existing `pvlib.pvystem.v_from_i` I get a value of 0.18443643025668521; all other values are exactly the same as you report. Whatever the mistake, it is causing the large residual.

thunderfish24, 2018-02-13:
@cwhanse Thanks for checking. I cannot figure out why I'm computing these anomalous values. I have opened #426 so that others can run the same exact script and compare output, including comparison of python (3.6.4) and numpy (1.14.0) versions.

cwhanse, 2022-10-05:
@pvlib/pvlib-maintainer expect a pull request in the near future from @reepoi that should close this issue.

PR #1573 (2023-03-15): Test pvsystem.singlediode against precisely generated IV curves

 - [x] Closes #411
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - ~~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/reference) for API changes.~~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Include [... cut]
closed 2023-03-15 by kanderso-nrel