pvlib/pvlib-python #413: Unable to use CEC modules in package overview example

grshennan, 2018-01-31:
Hi,

Just working my way through the procedural example given in the package overview. I copied-and-pasted the code and it runs fine. Then I want to select my own modules that are in the list of CEC modules rather than the Sandia modules

`sandia_modules = pvlib.pvsystem.retrieve_sam('cecmod')`
and
`module = sandia_modules['Seraphim_Solar_System_SRP_275_6PA']`

but when I do this I get an error, no matter which module I use. (I haven't written code to try them all but I have tried several and they all seem to produce the same error.)

```
C:\Users\Vignesh\Anaconda3\lib\site-packages\pvlib\clearsky.py:126: RuntimeWarning: invalid value encountered in fmax
  ghi = cg1 * dni_extra * cos_zenith * tl / tl * np.fmax(ghi, 0)
C:\Users\Vignesh\Anaconda3\lib\site-packages\pvlib\clearsky.py:131: RuntimeWarning: invalid value encountered in fmax
  bnci = dni_extra * np.fmax(bnci, 0)
Traceback (most recent call last):

  File "<ipython-input-27-d0a44f15ba73>", line 1, in <module>
    runfile('//emtserver/sharing/User Shared Folder/a. ENERGY/Glen/PVlib/package_overview_raw.py', wdir='//emtserver/sharing/User Shared Folder/a. ENERGY/Glen/PVlib')

  File "C:\Users\Vignesh\Anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 705, in runfile
    execfile(filename, namespace)

  File "C:\Users\Vignesh\Anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 102, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "//emtserver/sharing/User Shared Folder/a. ENERGY/Glen/PVlib/package_overview_raw.py", line 71, in <module>
    am_abs, aoi, module)

  File "C:\Users\Vignesh\Anaconda3\lib\site-packages\pvlib\pvsystem.py", line 1565, in sapm_effective_irradiance
    F1 = sapm_spectral_loss(airmass_absolute, module)

  File "C:\Users\Vignesh\Anaconda3\lib\site-packages\pvlib\pvsystem.py", line 1460, in sapm_spectral_loss
    am_coeff = [module['A4'], module['A3'], module['A2'], module['A1'],

  File "C:\Users\Vignesh\Anaconda3\lib\site-packages\pandas\core\series.py", line 601, in __getitem__
    result = self.index.get_value(self, key)

  File "C:\Users\Vignesh\Anaconda3\lib\site-packages\pandas\core\indexes\base.py", line 2491, in get_value
    raise e1

  File "C:\Users\Vignesh\Anaconda3\lib\site-packages\pandas\core\indexes\base.py", line 2477, in get_value
    tz=getattr(series.dtype, 'tz', None))

  File "pandas\_libs\index.pyx", line 98, in pandas._libs.index.IndexEngine.get_value

  File "pandas\_libs\index.pyx", line 106, in pandas._libs.index.IndexEngine.get_value

  File "pandas\_libs\index.pyx", line 154, in pandas._libs.index.IndexEngine.get_loc

  File "pandas\_libs\hashtable_class_helper.pxi", line 1210, in pandas._libs.hashtable.PyObjectHashTable.get_item

  File "pandas\_libs\hashtable_class_helper.pxi", line 1218, in pandas._libs.hashtable.PyObjectHashTable.get_item

KeyError: 'A4'
```

jkfm, 2018-01-31:
Hi grshennan,

the procedural example uses the SAPM model to calculate power output for solar modules. This model requires parameters supplied only in the Sandia module database. If you want to use modules from the CEC database you would need to use a different model, such as the SingleDiode model.

I would suggest you take a look at the [ModelChain documentation](http://pvlib-python.readthedocs.io/en/latest/modelchain.html#A-simple-ModelChain-example). There, you can specify "singlediode" for dc_model when initializing the ModelChain object.

wholmgren, 2018-01-31:
Thanks @jkfm

closed 2018-01-31 by wholmgren

grshennan, 2018-01-31:
Brilliant. Thank you.

This is a great project btw. I'm very new at this and not much of a programmer but I'll do what I can to chip in.

wholmgren, 2018-02-01:
Sounds good to me. I would suggest catching and reraising a KeyError with the suggestion to check the other database. I think trying to do more than that might be tricky.

Dod1samaz1ng, 2022-11-20:
> Hi grshennan,
>
> the procedural example uses the SAPM model to calculate power output for solar modules. This model requires parameters supplied only in the Sandia module database. If you want to use [...]

Has anoyone built an example of a procedure for this? I have been playing around and do not see a 1:1 relationship between the models. What is the level of effort for performing this?

This is for a course project and I am gauging the hours required for the group to work on this.