pvlib/pvlib-python #414: large tolerances in test_pvsystem.test_singlediode_floats_ivcurve

mikofski, 2018-01-31:
the absolute tolerance to compare test and expected output is 3 in [`test_pvsystem.test_singlediode_floats_ivcurve`](../blob/master/pvlib/test/test_pvsystem.py#L657) which allows almost any values to pass:

_e.g._:
```python
>>> out['i_sc'], expected['i_sc']
(6.9651723221583133, 6.96467476139632)
>>> np.isclose(out['i_sc'], expected['i_sc'])
False
>>> np.isclose(out['i_sc'], expected['i_sc'], atol=3)
True
>>> out['v_oc'], expected['v_oc']
(8.1063001465863103, 8.106300146586308)
>>> out['v_mp'], expected['v_mp']
(6.2243393757884284, 6.221535886625464)
>>> out['i_mp'], expected['i_mp']
(6.1362673597376753, 6.13902517979357)
>>> out['i'], expected['i']
(array([  7.00000000e+00,   6.53439455e+00,   3.48054918e-14]),
 array([  6.96517200e+00,   6.75588200e+00,   2.57571700e-14]))
>>> out['v'], expected['v']
(array([-0.7       ,  5.5749351 ,  8.10630015]),
 array([ 0.     ,  4.05315,  8.1063 ]))
```

IMHO these tolerance should be closer to 1e-3 maybe, or maybe set `rtol`?

adriesse, 2018-01-31:
Looks like a serious typo!  I vote for ``rtol``.

wholmgren, 2018-01-31:
Yes. Probably best to merge a fix for this before merging #409. Do you want to make the PR? I can do a quick one this afternoon if you want to stay focused on your current pvlib work.

thunderfish24, 2018-01-31:
Nice catch!

PR #415 (2018-02-12): fix bad atol in singlediode test
 - [x] Closes #414 
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests must pass on the TravisCI and Appveyor testing services.
 - [x] Code quality and style is sufficient. Passes ``git diff upstream/master -u -- "*.py" | flake8 --diff`` and/or landscape.io linting service.
 - [x] Adds description and name entries in the appropriate `docs/sphinx/source/whatsnew` file for all changes.
closed 2018-02-12 by wholmgren