pvlib/pvlib-python #416: inconsistent broadcasting of outputs from singlediode methods

mikofski, 2018-02-01:
The issue was raised in [some comments](./410#issuecomment-361974051) in #410 and #409 that the rules for broadcasting outputs are inconsistent among the single diode methods

* `singlediode` will only broadcast the output if `photocurrent` is an sequence, array or series
    - float -> OrderedDict of float
    - array or sequence -> OrderedDict of array or sequence
    - Series -> DataFrame
* `i_from_v` will broadcast the smallest size of either `voltage`, `photocurrent`, or `resistance_shunt`
* `v_from_i` will broadcast the smallest size of either `current`, `photocurrent`, or `resistance_shunt`

In those same comments @cwhanse proposed:
> to expand singletons when any parameter input is a vector, and to fail (gracefully) if two parameters are vectors of unequal length

I agree with this idea, and both `i_from_v` and `v_from_i` already have some boilerplate code to address this (note this snippet is lifted from `v_from_i`, with variable names suited for that function, change the
arguments to other methods like singlediode or mpp):

        # find the right size and shape for returns
        args = (current, photocurrent, resistance_shunt)  <-- list the args allowed to broadcast here
        size = 0
        shape = None
        for n, arg in enumerate(args):
            try:
                this_shape = arg.shape
            except AttributeError:
                this_shape = None
                try:
                    this_size = len(arg)
                except TypeError:
                    this_size = 0
            else:
                this_size = sum(this_shape)
                if shape is None:
                    shape = this_shape
            if this_size > size and size <= 1:
                size = this_size
                if this_shape is not None:
                    shape = this_shape
                    # there is no check for consistent shape
            else:
                # raise an exception here if you want
                pass  # do nothing to choose the smallest size

Now knowing the `size` and/or the `shape` one can let [`numpy.vectorize`](https://docs.scipy.org/doc/numpy/reference/generated/numpy.vectorize.html) cast the output as proposed:

        if size <= 1:
            V = v_from_i_fun(current, photocurrent, saturation_current,
                             resistance_series, resistance_shunt, nNsVth)
            if shape is not None:
                V = np.tile(V, shape)  <-- hacky way to make either array(V), array([V]), array([[V]]), ...
        else:
            vecfun = np.vectorize(v_from_i_fun)  <-- let numpy do the broadcasting for us!
            V = vecfun(current, photocurrent, saturation_current,
                       resistance_series, resistance_shunt, nNsVth)
        if np.isnan(V).any() and size <= 1:
            V = np.repeat(V, size)
            if shape is not None:
                V = V.reshape(shape)

On a related note: @cwhanse I realize now I misread [your comment](./410#issuecomment-362112394) and made `i_from_v` and `v_from_i` fail _silently_ instead of _gracefully. This might cause some confusion, in the short term, but maybe we can address it in this issue here?

@wholmgren is there a `PVLibException` class? What exception would you expect here, maybe `ValueError('All arguments must be either scalar or the same size.')`?

cwhanse, 2018-02-01:
I would prefer to fail verbosely and see an error message when the code for finding output shape etc. is presented with incompatible inputs.  The code is going to fail at some point but it's a challenge for many users to walk back though the stack dump and see what they did wrong.

mikofski, 2018-02-01:
✔️ addressed in f976f61 - fails verbosely if can't broadcast all inputs for `i_from_v` or `v_from_i`

thunderfish24, 2018-02-01:
I'm curious if `numpy.broadcast_arrays()` gives errors that make sense to a user when the inputs don't jibe well together?

mikofski, 2018-02-01:
IMO the error it raises is terse, but not opaque:
>ValueError: operands could not be broadcast together with shapes (3,) () () (2,) () ()

thunderfish24, 2018-02-01:
Oh Python, how I spend too much of my life sorting out your dynamic types!

wholmgren, 2018-02-12:
I don't fully understand this issue.

``singlediode`` returns multiple kinds of data. The pvlib style is to return labeled data in these situations. That means a DataFrame when the returned data is 2D and the input is a Series or DataFrame; OrderedDict if not. ``singlediode`` handles at least some kinds of numpy 2D+ inputs that I've used just fine. So, my experience with ``singlediode`` is that it already handles broadcasting for reasonable inputs. Maybe someone posted specific example of where it didn't do what you expected and I missed it in the flurry of discussion. Either way, could you (re)post it here?

``i_from_v`` and ``v_from_i`` return one kind of data each, so they don't need the labeling machinery. I am surprised to read that they broadcast to the smallest of ``voltage``, ``photocurrent``, or ``resistance_shunt``. That sounds like a bug. Perhaps that's due to something about ``np.broadcast_arrays`` that I don't understand (I don't have any experience with this function). Again, reproducible examples in this issue would help me to keep track of the status.

mikofski, 2018-02-15:
Sorry @wholmgren I think this is specific to #409 proposed methods that didn't seem to be broadcasting correctly?

I've tested the existing methods in the master branch with different size inputs and they seem to broadcast okay, as long as numpy arrays are used. There does seem to be some preferential treatment for `photocurrent` as it can be a list or an array.

Here are some examples:
#### `singlediode` 
* list of photocurrents, all other inputs scalar - works!
    ```singlediode([6, 7, 8], 1e-7, 0.001, 10, 1.23*0.026*72)```
* same size arrays of photocurrents and shunt resistance, others scalar - works!
    ```singlediode(np.array([6, 7, 8]), 1e-7, 0.001, np.array([10, 11, 12]), 2.3)```
* array of shunt resistance, all others scalar - works!
    ```singlediode(6, 1e-7, 0.001, np.array([10, 11, 12]), 2.3)```
* same size arrays of shunt and series resistance, others scalar - works
    ```singlediode(6, 1e-7, np.array([0.001, 0.002, 0.003]), np.array([10, 11, 12]), 2.3)```

#### `i_from_v`
* list of photocurrents, all other inputs scalar - works!
    ```i_from_v(10, 0.001, 2.3, 0.5, 1e-7, [6, 7, 8])```
* same size list of voltages and photocurrents, others scalar - works!
    ```i_from_v(10, 0.001, 2.3, [0.4, 0.45, 0.5], 1e-7, [6, 7, 8])```
* list of voltages, others scalar - works!
    ```i_from_v(10, 0.001, 2.3, [0.4, 0.45, 0.5], 1e-7, 6)```
* same size array of Rsh and voltages, others scalar - works!
    ```i_from_v(np.array([10, 11, 12]), 0.001, 2.3, np.array([0.4, 0.45, 0.5]), 1e-7, 6)```

I've change the title to make it specific to PR #409

thunderfish24, 2018-02-15:
I did add various combinations like these to the v_from_i() and i_from_v() unit tests when I added ideal device support. It’s certainly not exhaustive though.

cwhanse, 2024-10-30:
@mikofski can you confirm that this issue is closed by these lines: for [Bishop88 with newton](https://github.com/pvlib/pvlib-python/blob/b9946b236d49e43335935d2bb0e3c4b38b997757/pvlib/singlediode.py#L656); [here](https://github.com/pvlib/pvlib-python/blob/b9946b236d49e43335935d2bb0e3c4b38b997757/pvlib/singlediode.py#L680) and [here](https://github.com/pvlib/pvlib-python/blob/b9946b236d49e43335935d2bb0e3c4b38b997757/pvlib/singlediode.py#L757) for LambertW methods.

Given how long this has been open and the absence of other complaints about broadcasting, I'm inclined to just close this.

closed 2024-11-05 by cwhanse