pvlib/pvlib-python #421: bifacial PV system modeling

cwhanse, 2018-02-12:
implement NREL's rear-surface irradiance model for regular, fixed tilt arrays of bifacial PV models.

adriesse, 2018-02-12:
Is this your practice problem, Cliff?!?

cwhanse, 2018-02-12:
uh no this will be a big lift :)  #423 is a good practice problem.

cwhanse, 2018-02-12:
Mapping out how a fixed tilt bifacial PV system would be represented, here's a summary of my approach:

1. System description : add attributes to PVSystem for array racking geometry: number of array racks, distance between racks, height of rack above the ground, number of module rows on each rack, orientation of modules on the rack (landscape or portrait), spacing between modules (probably the rack area occupied by a module, rather than a distance between module edges), fraction of rear-surface module area 'obscured' by racking (this will be a derate on the rear irradiance), DC optimizers (module or cell level).  Perhaps we should use a new class as a child of PVSystem to contain these attributes?
2. A new database for bifacial module parameters that support only PVWatts. Bifacial modules have unique characteristics that aren't present in the SAPM or CEC datasets.  We have only measured a few bifacial modules so we're mostly providing a pattern to follow, not a dataset.  Bifacial parameters include: bifaciality  (ratio of rear to front power), temperature coefficient of power, rated power, length and width dimension, transparency factor (the fraction of front side irradiance that can pass through the module), cell array (number along length, number along width).
3. Irradiance functions. Two primary functions, one for rearside irradiance, and one for total effective irradiance. The rearside irradiance returns the rearside irradiance for each row of cells along the vertical dimension of the rack. This is a port and cleanup of the NREL bifacialvf code. 
4. A total irradiance function which returns the per-cell-row irradiance, front + rear, modified for 1) bifaciality factor, 2) soiling, 3) reflection, 4) rear-side shading, and 5) spectral effects.
5. A bifacial power function which calculates per-cellrow power from total irradiance and temperature, using PVWatts - I believe this is already `pvwatts_dc`.
6. A string power function which operates on per-cellrow power, DC optimizer settings, and string geometry (which I haven't figured out) to return power per string of modules.  I think this is in place of `scale_voltage_current_power` but I'm not clear on this point.

Interested in your thoughts.

wholmgren, 2018-02-13:
Regarding each numbered item:

1. Many of those racking parameters could be useful for (unimplemented) features for standard PV systems.
2. Easy to add a database to the pvlib/data directory.
3. I'm not familiar with the bifacialvf code, but that generally makes sense. Thinking dimensions... D_out = D_in x N_rows ?
4. Are the soiling and spectral effects unique to the bifacial nature of the system? I can imagine that they are. 
5. Yes, it sounds like you could call ``pvwatts_dc`` on your D_in x N_rows array. 
6. Maybe you need some combination of summing along the output array and using ``scale_voltage_current_power``?

I do think a new subclass of ``PVSystem`` would be useful for standardizing the process. It would override the ``PVSystem.get_irradiance`` method and implement new methods specific to the bifacial functions.

``pvsystem.py`` is already a big module. I wonder if we'd be better off putting most of the new stuff in e.g. ``bifacial.py``.

``ModelChain`` can probably accommodate this with very few changes.

cdeline, 2018-02-15:
Thanks everyone - are you planning to work off the existing Python code in the github.com/NREL/bifacialvf repository?

cwhanse, 2018-02-15:
Yes, but substantially refactored to integrate with the rest of pvlib, for transparency and hopefully less maintenance risk.

wholmgren, 2018-02-15:
@cdeline thanks for sharing -- I didn't realize it was on github. Dare I ask what the license is? I am not a lawyer, but if the license is not BSD 3 compatible then I think we need to stick to working off of papers or similar instead of from code.

adriesse, 2018-02-15:
Cool, I wasn't aware of it either.  I suspect the pvlib version will look different enough that copyright will not be a problem.

cdeline, 2018-02-15:
Will,  we are using BSD-3 licenses as well, and received an open-source designation from DOE so there should be no problems with porting the software into pvlib.

I don’t yet have our license document finalized and uploaded, but your question prodded me to make some progress on this!

Chris Deline
National Renewable Energy Laboratory
(303) 384-6359

cwhanse, 2018-02-15:
I should have asked first, thanks @wholmgren for bringing this up.  @cdeline good to here.  We'll keep prominent comments maintaining attribution and the supporting reference papers.

wholmgren, 2018-02-15:
@cdeline thanks for clarifying!

djaberpy, 2018-02-16:
cwhanse is right. Updating bifacial PV system modeling to PVlib will be a great job. I have been working on this topic recently, I can contribute and there is a lot to learn about bifacial systems and how it differs from standard PV. The issue of bifacial energy yield prediction is of interest nowadays.

cwhanse, 2018-06-20:
@anomam There isn't an open PR. I have a conversion of NREL's bifacialvf.py package to pvlib about 50% done. There's still substantial work left on that project.  I glanced through the pvfactors package. Two concerns we should discuss as we consider a merge into pvlib, or another path to more closely linking pvfactors with pvlib:

1.

mikofski, 2018-06-20:
Curious, why convert [NREL bifacialvf](https://github.com/nrel/bifacialvf) to pvlib? IMHO it would be better to leave it as an external dependence if possible. Ditto for other packages. 
Imo Easier to maintain IMO.

cwhanse, 2018-06-20:
In my opinion (but I've heard agreement from others) pvlib would benefit from an integral bifacial component (I could use it for our work at the lab). The NREL code will be difficult to maintain in its current state, and isn't really interfaced well with pvlib.  So we thought it worth the effort to migrate.

mikofski, 2018-06-20:
Ok

anomam, 2018-06-21:
@cwhanse thanks a lot for the replies. To answer your questions:

1. geopandas is indeed the underlying structure that's used to build the "registry", where the inputs/outputs and geometries are stored during the calculations. But it's used essentially as a convenient open-source wrapper around pandas that integrates ``shapely`` geometries, the real core package of the model. Originally when I created the algorithm, I was using simple dictionaries and lists. I'm sure there are much smarter and more efficient ways to do this, and it wouldn't be very difficult to change the nature of the registries, and use something more native to Python.

2. you're very right about the overhead though. Using ``shapely`` geometries makes building the PV geometries and calculating the view factors very intuitive and easy to maintain: since the 2D geometry is handled by the package, the view factor formulas look like [these](http://www.thermalradiation.net/sectionc/C-5.html), and you can visualize the PV array and shades on a plot at any point. Besides, the way ``pvfactors`` accounts for reflections is through a linear system of equations for all the surfaces in the PV array. 
But the cost for this is the computation speed: on my machine (i7 processors) it takes about 1 min and a half for 8760  points, and ~50s on a better machine when I was running batch simulations. Apart from model parametrization (which works very well btw), I haven't spent time thinking about replacing ``shapely`` with something that would make the computation faster.

Speed is a top priority for me as well. And maintainability and simplicity are big ones too . 
I was also planning on looking at ``bifacialvf`` because I heard it can run in 4 or 5 seconds, so I'd like to offer some help and contribute to the conversion of that package if that can accelerate its implementation in pvlib.
Let me know!

wholmgren, 2018-06-21:
@anomam we can add models to pvlib that depend on external packages. For pvfactors, I'd recommend making it an optional dependency, but I'll have to look at your package more carefully if we want to pursue this.

No reason we can't implement/wrap several bifacial algorithms in pvlib if there is community interest. We have multiple clear sky algorithms, transposition models, solar position models, pv models, etc.

cwhanse, 2018-06-21:
@anomam 2 minutes for an 8760 is OK for me, for a bifacial simulation. If we can not have the geopandas dependency I think in the long run we'll be happier, since that package is low versioned and brings a chain of dependencies, I worry about maintaining the connection.

I'd say go ahead with preparing a pull request. I really like the use of `shapely` and I think it will open possibiltiies for pvlib we haven't yet considered.

I will learn how to create a repository and get my partial translation of `bifacialvf` up, and I'd welcome help getting this project finished.

anomam, 2018-06-25:
@cwhanse @wholmgren Thanks a lot for the feedback!
Agreed, I will try to come up with something better than geodataframes for pvfactors before opening a PR.

anomam, 2018-08-01:
Hi @cwhanse @wholmgren , I removed the ``geopandas`` dependency in ``pvfactors``, but I haven't made any new release yet. I was hoping to get additional feedback from you if possible to integrate your requests into the next release before opening a PR in ``pvlib``. Is there any chance you could spare some time looking at the [pvfactors package](https://github.com/SunPower/pvfactors) and let me what changes you would like to see? Any feedback would be much appreciated

cwhanse, 2018-08-01:
@anomam happy to.

PR #635 (2019-01-22): pvfactors limited implementation for bifacial calculations
pvlib python pull request guidelines
====================================

Thank you for your contribution to pvlib python! You may delete all of these instructions except for the list below.

You may submit a pull request with your code at any stage of completion.

The following items must be addressed before the code can be merged. Please don't hesitate to ask for help if you're unsure of how to accomplish any of the items below:

 - [x] Closes  #421
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthed [... cut]
closed 2019-01-22 by wholmgren