pvlib/pvlib-python #422: irradiance.globalinplane appears redundant with total_irrad

cwhanse, 2018-02-12:
Suggest removing globalinplane.

wholmgren, 2018-02-12:
I think they're different, but I'm struggling to think of a good use case for ``globalinplane``.

``globalinplane`` computes POA global from modeling intermediates (AOI, DNI, POA diffuse). 

``total_irrad`` computes POA components from more basic information (system specs, solar position, DNI, GHI, DHI).

jforbess, 2018-02-13:
I am mixed on this, because I have used both, but not necessarily for a
good reason, just because of whichever tutorial I was copying at the time.

I think I prefer globalinplane because I occasionally have need to modify
the intermediates rather than use the default assumptions. Also, does
total_irrad work with tracking? I can't remember off the top of my head.

wholmgren, 2018-02-13:
``total_irrad`` accepts scalar (fixed) or vector (tracking) surface azimuth and tilt. It does not automatically compute tracking angles, though.

My post above was a bit misleading. I suggested that ``globalinplane`` returns only POA global, but it actually returns POA global, POA direct, and POA diffuse (total). We could probably come up with a better name for ``globalinplane``. Something like ``poa_components``. 

Thinking through this a little more... ``total_irrad`` does 3 things:

First, ``total_irrad`` provides a unified interface to the various diffuse sky models. This aspect could be refactored into a separate function (``get_diffuse``?) and used within ``total_irrad``.

Second, ``total_irrad`` calls the ``grounddiffuse`` function.

Third, ``total_irrad`` assembles the POA components into a data structure (DataFrame or OrderedDict). ``total_irrad`` could be refactored to use a renamed ``globalinplane`` internally. The renamed ``globalinplane`` would need to include the sky diffuse and ground diffuse to its output for API consistency.

So, ``total_irrad`` would be reduced to a wrapper for the 3 steps described above. It would not do any calculations itself.

While we're at it, I'd welcome suggestions for a better name than ``total_irrad``.

cwhanse, 2018-02-13:
I think those ideas are good, reduce total_irrad to a wrapper, and refactor the wrapped functions.  My instinctive reaction to 'total_irrad' is 'get_irradiance_components'

I like function names that start with verbs.

PR #427 (2018-08-14): add get_[relative_airmass,sky_diffuse,ground_diffuse,extra_radiation,total_irradiance] and pvlibDeprecationWarning
 - [x] Closes #422 
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests must pass on the TravisCI and Appveyor testing services.
 - [x] Code quality and style is sufficient. Passes ``git diff upstream/master -u -- "*.py" | flake8 --diff`` and/or landscape.io linting service.
 - [x] New code is fully documented. Includes sphinx/numpydoc compliant docstrings and comments in the code where necessary.
 - [x] Updates entries to ` [... cut]