pvlib/pvlib-python #434: Return sky diffuse irradiance components in sky diffuse dataframe

cwhanse, 2018-02-22:
`irradiance.perez` can return isotropic, circumsolar and horizon components of sky diffuse irradiance if `return_components = True`.  Currently, if `return_components` then results are returned as a tuple `(sky_diffuse, diffuse_components)` with the elements being a pandas.Series and pandas.Dataframe, respectively.

I propose that we always return a Dataframe with column `sky_diffuse` and optionally columns for `isotropic, circumsolar, horizon`.  This avoids having the output of the perez model of a different type (tuple instead of pandas object) and having to unpack the tuple when diffuse_components are requested.

wholmgren, 2018-02-22:
This is an interesting idea. I'm concerned about maintaining API compatibility across all of the transposition models. I think the default behavior of all of the models should be the same, if possible. So, I'm reluctant to default to returning a one-column DataFrame from ``perez`` while the others return Series. 

We could make similar changes to all of the models. I don't know if that would be useful or busy work. What would the keys look like for each function? For example

``isotropic`` returns ``sky_diffuse, isotropic``?
``haydavies`` returns ``sky_diffuse, isotropic, circumsolar``?
``perez`` returns ``sky_diffuse, isotropic, circumsolar, horizon``?

cwhanse, 2018-02-22:
Good point about compatibility.  We could return a Series if `return_components = False` rather than a one column data frame.  The hiccup for me is dealing with the tuple instead of a pandas object.

We could propagate the component capability into the other sky diffuse models, although perez is likely to be the most frequently used for this purpose.

wholmgren, 2018-02-22:
Sure, that sounds like a quick fix then. 

For reference, the ``return_components`` feature was added in #259. It does not appear that we discussed the idea of returning a single DataFrame with the total and the components.

wholmgren, 2018-09-06:
@cwhanse I suggest that we limit the scope of this particular issue to ``irradiance.perez`` and create new issues if want to expand this feature to new transposition models.

PR #565 (2018-09-07): change irradiance.perez(return_components=True) to return dict/DataFrame
 - [x] Closes  #434
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/contributing.html).
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests (usually) must pass on the TravisCI and Appveyor testing services.
 - [x] Updates entries to `docs/sphinx/source/api.rst` for API changes.
 - [x] Adds description and name entries in the appropriate `docs/sphinx/source/whatsnew` file for all changes.
 - [x] Code quality and st [... cut]
closed 2018-09-07 by wholmgren