pvlib/pvlib-python #435: Artefact in Ineichen clearsky model for sun close to horizon

kdebrab, 2018-02-23:
Consider following example:
```python
import pandas as pd
import pvlib
times = pd.date_range('2018-02-16', '2018-02-17', freq='min')
location = pvlib.location.Location(latitude=60., longitude=0.)
for model in ['ineichen', 'haurwitz', 'simplified_solis']:
    location.get_clearsky(times, model=model)['ghi'].plot(label=model, legend=True)
```
which results in following figure:
[image]

The (default) 'ineichen' clearsky model seems erratic at sunrise and sunset. The effect is especially visible for high altitudes during Winter when the sun slowly approaches the horizon.

Upon closer look in the code, I found the following comment:
https://github.com/pvlib/pvlib-python/blob/04bd7532f54b550313d84309c3d3d19818694ccc/pvlib/clearsky.py#L98-L103
which refers to the formula:
https://github.com/pvlib/pvlib-python/blob/04bd7532f54b550313d84309c3d3d19818694ccc/pvlib/clearsky.py#L122-L126

However, apart from the typo mentioned, there is another important difference between both formula of Solar Energy 73, and that's the `np.exp(0.01*airmass_absolute**1.8)` factor, which is only present in the formula of pg 311, but not in the formula of pg 156!

It seems like the artefact is exactly a result of that `np.exp(0.01*airmass_absolute**1.8)` factor. This factor slowly increases from 1.01 at solar zenith to 1.95 at a zenith angle of 85°, but then shoots to 1000 when the sun reaches the horizon.

So, maybe there was an error also in the formula of Solar Energy 73, pg 311, and that factor wasn't supposed to be there?

Nevertheless, in reference [3] (Reno et al., 2012), this model was found to excel for US sites with the same formula as used by pvlib. The US sites were all below the [49th parallel](https://en.wikipedia.org/wiki/49th_parallel_north). I wonder whether this model is still recommendable for higher latitudes?

kdebrab, 2018-02-23:
There is actually another version of the paper of Ineichen, which doesn't contain the typo nor the factor causing the artefact: http://citeseerx.ist.psu.edu/viewdoc/download?doi=10.1.1.535.4248&rep=rep1&type=pdf.

adriesse, 2018-02-23:
Hi Karel!  I saw this effect too but simply didn't have the time to investigate and report--so thank you for this!  One hypothesis I had was that this might have worked ok for hourly averages.

kdebrab, 2018-02-23:
Hi Anton, The effect indeed mostly disappears in hourly averages.

Unless you go even further north. Then the whole curve gets really biased. See e.g. the result for 70° latitude:
[image]

wholmgren, 2018-02-23:
Thanks for the detailed report. This artifact and the discrepancies among the papers and code is part of what motivated me to add the simplified solis model to the library.

adriesse, 2018-02-23:
Time to write an email to Pierre Ineichen?

mikofski, 2018-02-24:
Hi @kdebrab thanks for catching this. How does Ineichen compare to haurwitz and solis when the mystery factor `np.exp(0.01*airmass_absolute**1.8)` is removed? And how do they compare to [Bird](http://pvlib-python.readthedocs.io/en/latest/generated/pvlib.clearsky.bird.html)?

adriesse, 2018-02-28:
Pierre will retire in May, so don't wait too long to ask!

mikofski, 2018-02-28:
Gasp!

adriesse, 2018-04-17:
I checked with Pierre, and he explained that Richard Perez added the exponential term in the later paper based on further analysis of measurements at 10 continental US sites.  I guess no one checked the effect on locations outside this geographic area.

I would suggest that function be modified to accept a flag to indicate whether the Perez enhancement is to be used or not, with a default value of False.

wholmgren, 2018-04-17:
Thanks for checking. I support your proposed solution. I'm not going to get to it anytime soon, though, and would welcome a pull request.

cwhanse, 2018-04-17:
I concur with @adriesse proposal.  I would add a comment to the function signature warning that the Perez enhancement may produce biased results at high latitude, and link the False/True condition for the flag to the appropriate reference.

closed 2018-08-06 by cwhanse

martinherrerias, 2020-09-22:
One way to remove the artifacts while introducing less bias is to clip the `airmass_absolute` values to the point where the exponential correction becomes problematic. Something like:

`airmass_critical = (cg2*(fh1+fh2*(tl-1))/0.018).^1.25`
`airmass_clipped = np.fmin(airmass_absolute, airmass_critical)`
`ghi = (np.exp(-cg2*airmass_clipped*(fh1 + fh2*(tl - 1)) + 0.01*airmass_clipped**1.8))`
`...`
`ghi = cg1 * dni_extra * cos_zenith * tl / tl * np.fmax(ghi, 0)`

`airmass_critical` is a minimum for the combined exponential terms, which shoot up afterwards. Loosely:
`airmass_critical = argmin(f(x)), for f = exp(-cg2*x*(fh1 + fh2*(tl - 1)) + 0.01*x^1.8))` 

This threshold starts at AM = 2.6 (for turbidity = 1 and altitude = 0), equivalent to ~22.5° solar elevation, and goes up to O(10) for higher turbidities and altitudes. For most cases, it corresponds to solar elevation angles below the 4° cutting threshold of the CIE 1994 quality control guidelines that Ineichen & Perez seem to have used, i.e. they probably didn't see beyond it.

Clipping at the minimum makes the adjustment smooth (C1), and leaves the results of the model untouched for most operating conditions. 

[image]

cwhanse, 2020-09-22:
@martinherrerias your proposal makes a lot of sense to me, now that I better understand the the role that correction plays. Rather than continuing in this old issue, would you like to open a new issue? If you are also inclined a pull request is welcome.

adriesse, 2020-09-22:
I would suggest controlling this new feature by the same flag or a new one. It should be possible to run the model as published.

This reminds me of some discussions about using models outside of the range of inputs for which they were developed and/or validated.

mikofski, 2020-09-22:
Are there any references for this new feature? @martinherrerias any plans to present or publish this idea? I think that would be useful. Thanks!

PR #459 (2018-08-06): Add option to control airmass/perez enhancement factor in clearsky.ineichen
See #435 for context.

 - [x] Closes #435
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests must pass on the TravisCI and Appveyor testing services.
 - [x] Code quality and style is sufficient. Passes ``git diff upstream/master -u -- "*.py" | flake8 --diff`` and/or landscape.io linting service.
 - [x] New code is fully documented. Includes sphinx/numpydoc compliant docstrings and comments in the code where necessary.
 - [x] Updates entries to `docs/sphinx/sourc [... cut]