pvlib/pvlib-python #441: Does PVLib cover exceedence probabilities?

obeavers, 2018-03-25:
Hey guys,

Just found this project -- awesome job so far.

I'm looking at the examples in the documentation overview where annual yield is calculated using the basic chain, and was wondering if there was any estimate on what the exceedence probability would be for the models being called/applied?

Thanks.

cwhanse, 2018-03-26:
I'll assume that you are asking about exceedance probabilities for annual energy. Those exceedance probabilities are usually based on inter-annual variability in weather.  If you have multiple years of weather, you can calculate annual energy for each year, form the empirical distribution and estimate the percentiles.

wholmgren, 2018-03-27:
I agree with the above responses. Let's take any further discussion to the mailing list or stack overflow. We try to keep the issue tracker focused on problems/enhancements to the code. See "Getting support" for more: http://pvlib-python.readthedocs.io/en/latest/package_overview.html#getting-support