pvlib/pvlib-python #447: logging.basicConfig in __init__.py causes misbehavior after import

Gemmu, 2018-04-05:
Since `basicConfig` is called as a import side-effect of pvlib, any software that imports anything from pvlib can not use it's own basicConfig (if basicConfig was already called, any further call just does nothing - not even giving a warning).

This leads to unexpected behavior (no more debug output) which is rather hard to track down.

I don't know whether it is bad to copy issues word-by-word, but this affects pvlib as well: https://github.com/circus-tent/circus/issues/826

The problem can be reproduced using the code form Python logging cookbook:
https://docs.python.org/2/howto/logging-cookbook.html#using-logging-in-multiple-modules
Adding `import pvlib` will cause all logging to be ouputted to terminal.

Maybe this is by design, but it's a bit difficult to get around.

wholmgren, 2018-04-05:
I think all logging in pvlib can be removed, with maybe a couple of calls being replaced with warnings. Most pvlib logging is a holdover from my early efforts to remove print and some debugging.

mikofski, 2018-04-05:
@Gemmu if you set [`propagate`](https://docs.python.org/3/library/logging.html) to `False` then it shouldn't inherit the root logger.

@wholmgren instead of using `basicConfig` create a pvlib logger that has it's own handlers and formatters so it doesn't propagate. See the [HOWTO on configuring loggers](https://docs.python.org/3/howto/logging.html#configuring-logging) or the [logging cookbook](https://docs.python.org/3/howto/logging-cookbook.html#using-logging-in-multiple-modules)

wholmgren, 2018-04-05:
@mikofski sure, but I don't think pvlib should be logging anything. Do you find pvlib logging to be useful? I think issuing warnings and pushing developers towards pdb is more appropriate for the vast majority of the existing logging calls.

PR #448 (2018-04-09): replace logging.warning with warnings.warn, remove logging.debug
Will anyone miss these internal pvlib logging calls?

With these changes, ``grep logg pvlib/* -R`` returns nothing.

 - [x] Closes #447 
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests must pass on the TravisCI and Appveyor testing services.
 - [x] Code quality and style is sufficient. Passes ``git diff upstream/master -u -- "*.py" | flake8 --diff`` and/or landscape.io linting service.
 - [x] New code is fully documented. Includes sphinx/numpydoc compliant docstrings an [... cut]