pvlib/pvlib-python #449: DISC algorithm air mass

adriesse, 2018-04-11:
As far as I can tell from reading the original report, the air mass value required is _not_ the pressure-corrected, AMa.

cwhanse, 2018-08-07:
@adriesse you are correct that Maxwell et al report presents AM (air mass) as relative air mass (not adjusted for atmospheric pressure).  However, we based the Matlab function on NREL's coding of the DISC model [https://www.nrel.gov/grid/solar-resource/disc.html](url) which uses AM as absolute air mass.

We could change to relative air mass in `irradiance.disc` to remain strictly consistent with the publication, or leave as is (and add a reference) to remain consistent with NREL's code.  I can see arguments for either option.

wholmgren, 2018-08-15:
We should at least add a note to the ``disc`` documentation of the ``pressure`` keyword argument. We could also allow ``pressure=None`` to ensure that no correction is applied.

cwhanse, 2018-10-19:
I propose that we leave the DISC function as using absolute air mass, and enhance by allows `pressure=None` to turn off the conversion from relative to absolute air mass. Agree, or no?

adriesse, 2018-10-19:
Sure.  The current default value `pressure=101325` effectively already does this, but `None` would represent the situation better.  A sentence or two for posterity in the doc string is probably the most important.

PR #625 (2018-11-26): note and tests for disc relative vs absolute airmass
 - [x] Closes #449 
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/contributing.html).
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests (usually) must pass on the TravisCI and Appveyor testing services.
 - [x] Updates entries to `docs/sphinx/source/api.rst` for API changes.
 - [x] Adds description and name entries in the appropriate `docs/sphinx/source/whatsnew` file for all changes.
 - [x] Code quality and style is sufficient. P [... cut]
closed 2018-11-26 by wholmgren