pvlib/pvlib-python #457: Inverter with multiple MPPTs

jhfatehi, 2018-05-11:
It's been a couple years since I have used pvlib (and the progress is amazing) so forgive me if this has already been covered.  Is there an implementation of inverters with multiple MPPTs?

cwhanse, 2018-05-14:
@jhfatehi In a word, no. pvlib calculates system DC power by multiplying a module's DC output by number of modules per string and by strings per inverter. I'm sure one could find a way to use pvlib to model a  multi-MPPT inverter, but it probably requires defining multiple systems and adding the results outside of pvlib.

wholmgren, 2018-05-16:
Perhaps we could add an example to the documentation.

Also consider checking SAM/ssc. We could discuss reimplementing their solution, if available.

Peque, 2020-05-21:
@jhfatehi Did you workaround this somehow? If so, can you share how you did that?

@wholmgren I am guessing there is no example in the documentation? :innocent:

wholmgren, 2020-05-21:
No example, contribution welcome: python file in [docs/examples](https://github.com/pvlib/pvlib-python/tree/master/docs/examples) renders to [here](https://github.com/pvlib/pvlib-python/tree/master/docs/examples)

jhfatehi, 2020-05-21:
This was my workaround

dc_string1 = dc_string_out(string1_tilt, string1_azimuth, cec_module, string1_mods, solpos, dni_extra)
dc_string2 = dc_string_out(string2_tilt, string2_azimuth, cec_module, string2_mods, solpos, dni_extra)

dc_system = pd.DataFrame()
dc_system['p_mp'] = dc_string1.p_mp + dc_string2.p_mp
dc_system['v_mp'] = (dc_string1.p_mp*dc_string1.v_mp + dc_string2.p_mp*dc_string2.v_mp)/dc_system.p_mp

acs = pd.DataFrame()
acs['power'] = pvlib.pvsystem.snlinverter(dc_system.v_mp, dc_system.p_mp, sapm_inverter)

Peque, 2020-05-21:
@jhfatehi Thanks for sharing! :blush: 

Out of curiosity, may I know the inverter you were using? To have a look at its specs in the database. I am wondering if there is information there about the number of MPPTs for example.

jhfatehi, 2020-05-21:
The inverter model is reducing power depending on the inverters voltage/power/efficiency curve.  There is no MPPT happening at the inverter.  The ideal Vmp and Imp are calculated in the diode model.  The challenge with 2 MTTPs is what voltage to use.  In my example where I have 2 strings I used a weighted average of Vmp of the 2 string where I weighted by Pmp of the string.  I honestly have no clue if this is remotely accurate because I am not familiar with the topology of multiple MPPT inverters.  Another way to do it would be to treat each MPPT as a separate inverter but I image this would be optimistic for AC production because you likely will be operating at a higher efficiency point because you have roughly same voltage but lower power in each inverter.

To answer your question I was using a SolarEdge inverter which operates at a fixed voltage in real life.  The MPPT high and low voltage in the inverter models is for information only.  It does not impact the inverter efficiency calculation.

Peque, 2020-05-24:
@jhfatehi Wouldn't your weighted solution be the case for a single-MPPT inverter? Having 2 different strings connected in parallel to the single MPPT input would result in the MPPT resolving to a weighted operating point selection.

I mean, I thought having 2 MPPTs would allow the inverter to maximize the power on each MPPT input separately. So for your case with 2 strings I would have thought the inverter was able to handle them independently, as if you had two inverters.

Am I understanding something wrong? I guess for a 2-MPPT inverter there are not really 2 inverters inside the case, but maybe it had 2 DC-DC converters and a single inverter to allow separate power optimization for each input?

Peque, 2020-06-03:
I asked this question in Stack Exchange:

https://electronics.stackexchange.com/questions/503523/how-to-calculate-output-power-of-a-multi-mppt-inverter

In case anyone can help answering it. :blush:

cwhanse, 2020-10-26:
Some progress to report on this issue, which has emerged as a roadblock to #1067.

- an inverter with multiple MPPTs can be reasonably modeled with the available inverter models. The approach is to regard each MPPT input as producing a fraction of the inverter's AC output. The DC-AC inversion efficiency curve for a MPPT input is a scaled version of the DC-AC curve for the whole inverter; this is supported by the way inverter efficiency is measured (at least, by the CEC method for measurement). This is the default approach in [Pvsyst](https://www.pvsyst.com/help/) for modeling such inverters. I've got some technical notes to write up and will post to pvpmc.org.
- adding capability to pvlib to model multiple MPPT inverters will require a new function and also refactoring the existing `pvlib.inverter.sandia` function. In `sandia`, limiting inverter output to maximum Pac needs to be done after the AC power from each MPPT are combined. I don't think the `pvlib.inverter.sandia` signature should change. The new function will use the refactored parts of `sandia` with at least one new parameter for the fraction of power handled by each MPPT (default is 1 / number of MPPT), and perhaps other new parameters.
- inverters are not specified with limits on each MPPT in terms of power. Rather, DC current and voltage limits are specified, e.g., [here](https://files.sma.de/downloads/STPCORE1-DS-en-27.pdf?_ga=2.148681207.1330373776.1603726449-1399979901.1603726449). Multiplying these limits obtains a DC power much greater than the inverter AC rating divided by the number of MPPT inputs, so inverter manufacturers are designing these devices for very unbalanced operations (i.e. several arrays with very different DC output). The current/voltage limits by MPPT are on the spec sheets, but they are not in the CEC inverter data nor in the SAM file which pvlib uses. Pvsyst warns users that it is their responsibility to ensure that per-MPPT limits are respected.
- For pvlib, I think we should require users to supply the per-MPPT parameters which "clip" the DC input from each array, which is then inverted to AC using a scaled version of the Sandia model, added, then the sum is clipped to the inverter's AC maximum output.

cwhanse, 2020-10-27:
A proposed approach for modeling inverters with multiple MPPT. I welcome any feedback.
[Proposal to model multiple MPPT inverter.pdf](https://github.com/pvlib/pvlib-python/files/5448398/Proposal.to.model.multiple.MPPT.inverter.pdf)

Peque, 2020-10-30:
@cwhanse Thanks for sharing! :blush: I will definitely have a look at it in 1-2 weeks!

Do you know if there is plan to include an MPPT count parameter in the CEC database? :stuck_out_tongue_winking_eye:

cwhanse, 2020-10-30:
> Do you know if there is plan to include an MPPT count parameter in the CEC database?

Unfortunately, no. That database is out of pvlib's control. The file of inverter model parameters is obtained from NREL's SAM, and NREL's database is generated by applying the inverter model fitting algorithm to data in the CEC equipment list.

The CEC doesn't request the MPPT count (or voltage/current limits, or many other inverter parameters) so that data aren't included in the database that lies behind the file of inverter parameters. See the Inverter tabs [here](https://www.energy.ca.gov/programs-and-topics/topics/renewable-energy/solar-equipment-lists) for information on the CEC equipment lists.

PR #1085 (2020-12-02): implement multi-MPPT inverter model
 - [x] Closes #457
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :i [... cut]
closed 2020-12-02 by wholmgren