pvlib/pvlib-python #463: Implement CEC module performance model in place of desoto

cwhanse, 2018-05-16:
The Desoto module performance model and the CEC model differ in that the CEC model has one additional parameter (an adjustment to temperature coefficients for Isc and Voc).  pvlib-python implements only the Desoto model, but supplies parameters that are fit for the CEC model.  Replacing the Desoto model with the CEC model would remove this inconsistency.  The differences between the two model's results are quite minor.

adriesse, 2018-05-16:
Are you talking about the so-called "adjust" parameter, whose origins I had trouble tracing a year or more back?

cwhanse, 2018-05-16:
Yes

adriesse, 2018-05-16:
I retained the impression that it was used to influence the fitting process rather than being a model parameter...

cwhanse, 2018-05-16:
Adjust enters the IV curve calculation via the equation for light current: IL = G * ( IL0 + alphaprime (Tc - T0)), and alphaprime = alpha ( 1 - Adjust/100).  As far as I know this is only detailed in A. Dobos paper describing the 6 parameter model coefficient calculator. The procedure to determine the value for Adjust is still obscure to me despite Aron's paper.

adriesse, 2018-05-16:
The way I saw it is that your diode model parameters may end up with a new, adjusted alpha (compared to the datasheet value) but to me that doesn't really constitute an additional model parameter. 

There are a number of things that worry me about that paper, including the sanity check that allows alpha and beta to be adjusted by +/- 100%.

cwhanse, 2018-05-17:
We can (and I agree that we should) add the CEC model and keep the Desoto functions.  However, strictly speaking we have CEC model parameters, rather than Desoto parameters.  The values in SAM are determined simultaneously with the Adjust parameter (in the CEC bot not the Desoto model) and thus there is some (certainly minor) influence of the Adjust value on the other parameters.

thunderfish24, 2018-05-17:
FWIW there is https://www.nrel.gov/docs/fy18osti/67399.pdf, esp. section 10.4. There are many reasons why “the” reported Isc temperature coefficient could be need adjustment, such as its spectral dependence during measurement. Why the same adjustment would be used simultaneously on Isc and Voc (presumably in data fitting) I cannot say. I too would be suspect if the adjustment were large, and is it true then that an IV curve’s Voc could be very different from the “adjusted” value? If yes, then what Voc value would pvlib report?

cwhanse, 2018-05-17:
The Adjust parameter acts to bring the model's temperature coefficients for power and Voc into line with the datasheet values.  The temperature coefficients of power and Voc are model outputs rather than model parameter.  The CEC model implements Adjust to provide this matching, PVSyst uses a temperature coefficient for the diode factor.  In either case, the effect is to add a model parameter for fitting purposes.

wholmgren, 2018-05-17:
Are we in agreement that:

1. CEC model should be added to ``pvsystem.py`` as a function with a corresponding ``PVSystem`` wrapper method.
2. ModelChain should use CEC model if it detects that ``PVSystem.module_parameters`` comes from the NREL database (possibly via the presence/absence of an ``Adjust`` parameter).
2. Documentation should recommend using CEC model if module parameters come from the NREL database. Perhaps will be needed in a few places.
2. Documentation examples that currently use desoto should be modified to use CEC (they all come from the NREL database)

cwhanse, 2018-05-17:
#1 - 4 : I agree.

adriesse, 2018-05-17:
I'm not sure.  I think part of my unease comes from the frequent use of the word _model_ in this discussion without knowing exactly what it refers to in each instance.  I think there are three separate "calculation steps" (for lack of a better word): the one-diode I-V curve equation (which hasn't changed in a long time and isn't associated with any particular name); the process of determining its 5 parameters at reference conditions from various measured quantities; and the translation of its parameters to represent different operating conditions.  The third step is governed by additional measured quantities, such as alpha, or additional parameters produced in the second step, such as alphaprime/adjust.

Is there an easy way to keep these distinctions clear in the code structure etc?

cwhanse, 2018-05-17:
I try to be consistent with my own terms, and hope that others adopt them, but time will tell.  The single diode equation is a model (in a general sense) for an IV curve but it is not a model for a photovoltaic device. The single diode equation has 5 coefficients (I prefer not to call them parameters, although I've sort of given up on making that distinction).  When I write 'single diode model' I mean the diode equation together with the other equations describing how the 5 coefficients change with irradiance, temperature, etc.  The constants appearing in this larger set of equations are the single diode model parameters, e.g., light current at STC, alpha, diode factor, etc.  E.g., the Desoto model actually has 8 parameters: IL, Io, n, Rsh and Rs at STC (5), alpha_Isc, Eg at STC and dEgdT at STC. (3 more parameters). 

In this nomenclature, Desoto is a photovoltaic performance model, PVsyst includes a photovoltaic performance  model, but the single diode (or n-diode) equation is not a PV performance model.

To comment on what @adriesse writes, the 3 steps are separate and each is important. The '5 parameters at reference conditions'.  Those 5 values are some, but not all, of the parameters for the Desoto, CEC or PVsyst performance models.  To summarize what is in pvlib-python for each of the 3 steps mentioned by @adriesse :
1) determine model parameters from measurements,  No capability currently in pvlib-python but there's a stale, unfinished pull request for a method to estimate PVsyst model parameters.
2) given parameters that determine a single diode performance model, calculate the 5 coefficients for the single diode equation at the desired irradiance etc. conditions.  Here pvlib-python has calcparams_desoto, we could add calcparams_PVsyst, calcparams_CEC, etc.
3) calculate an IV curve given a set of 5 coefficients.  This is the current singlediode function.

adriesse, 2018-05-20:
Good, I don't think we disagree on anything important.

If in the proposed action 1 above, 'CEC model should be added' means making a `calcparams_CEC` function (or perhaps `calc_1d_params_CEC`) then I fully support it. 

The rest sounds sensible too.  Maybe the above discussion will turn out to be helpful when choosing words for the documentation updates.

thunderfish24, 2018-05-21:
With due consideration of a consistent API, I have no issue adding various model implementations. Personally, and following [this terminology, section 2.1](https://www.researchgate.net/profile/Jenny_Brynjarsdottir2/publication/278156450_Learning_about_physical_parameters_The_importance_of_model_discrepancy/links/55d499e408ae6788fa352029/Learning-about-physical-parameters-The-importance-of-model-discrepancy.pdf) with terms "calibration parameters" (e.g., dark current or shunt resistance at reference conditions) and "control variables" (e.g., temperature or irradiance). It seems that the term "coefficients" are indeed useful to distinguish values in the 5-parameter single diode equation that are computed as some function of calibration parameters and control variables in a PV performance model.

Am I correct that the adjust parameter only appears in the CEC's PV performance model in "the" short-circuit temperature coefficient, even although it might itself be calibrated using other "measured output values" such as various datasheet temperature coefficients?

Furthermore, when computing, say, a Voc value at a temperature different from reference conditions, can one expect that using an datasheet (or adjusted?) Voc temperature coefficient to be better/worse than computing the "same" value from the corresponding I-V curve from the PV performance model?

cwhanse, 2018-05-21:
> Am I correct that the adjust parameter only appears in the CEC's PV performance model in "the" short-circuit temperature coefficient, even although it might itself be calibrated using other "measured output values" such as various datasheet temperature coefficients?

Yes, Adjust is a calibration parameter in the CEC performance model, appearing in both the equation for the short circuit current temperature coefficient and the equation for the open circuit voltage temperature coefficient betaVoc.  However, betaVoc is a model output and is not used in the calculation of the IV curve.

> Furthermore, when computing, say, a Voc value at a temperature different from reference conditions, can one expect that using an datasheet (or adjusted?) Voc temperature coefficient to be better/worse than computing the "same" value from the corresponding I-V curve from the PV performance model?

I read your question as: how does Voc (from the single diode equation) compare to Voc computed with the linear expression involving betaVoc?  That's a good question.  I'll respond unhelpfully and say that the equation defining Voc as a model output is the single diode equation, so from the point of view of faithfully implementing a model in software, your question is moot.

adriesse, 2018-05-21:
I wondered about the second question too, and only have a theory, not an answer.

The median value for the _adjust_ parameter in the CEC database is about 10%.  This would seem to imply that the target beta for the CEC parameter optimization/search was 10% above the datasheet value, and consequently I would expect that when using this model I would see a higher beta value than the spec sheet.  

If this is theory correct, then I would worry a little bit about modules with large _adjust_ values.  Voc is important for system sizing.

thunderfish24, 2018-05-23:
While the implementation details of how CEC parameters are calibrated using _adjust_ are not entirely clear to me (and this approach seems fraught with likely issues), I agree that large _adjust_ values appear worrisome with respect to goodness of fit.

It also appears that the only difference between the CEC and Desoto models' calculations is use of the short-circuit temperature coefficient vs. the "adjusted" short-circuit temperature coefficient. Is that correct?

I suppose that a related question here is whether pvlib should compute various temperature coefficients (i.e., various derivatives w.r.t. temperature) at reference/arbitrary conditions.

cwhanse, 2018-05-23:
@thunderfish24 agree with you about the calibration. I suspect one reason for non-zero values of Adjust is that the calibration is matching the derivative at STC to an average rate of change over a range of temperature. You are correct, the only difference between CEC and Desoto that I know is the use of Adjust in the equation for the short-circuit current temperature coefficient.

We can certainly write functions to calculate derivatives for a specific single diode model, but I question whether there is widespread interest in them.

thunderfish24, 2018-06-05:
@cwhanse Do you know where the original CEC model is actually documented, esp. the "auxiliary equations"? I've been relying on [this SAM documentation](https://www.nrel.gov/docs/fy18osti/67399.pdf), but I'm not sure what changes may have been made here.

I have seen a significant difference in model calibration quality between the Desoto-style (no use) and PVsyst-style (use) implementation of the ideality factor in the auxiliary equation for the reverse saturation current (i.e., the reverse saturation current as a function of temperature).

PR #560 (2018-09-08): Implement CEC model
pvlib python pull request guidelines
====================================

Thank you for your contribution to pvlib python! You may delete all of these instructions except for the list below.

You may submit a pull request with your code at any stage of completion.

The following items must be addressed before the code can be merged. Please don't hesitate to ask for help if you're unsure of how to accomplish any of the items below:

 - [x] Closes  #463 
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/contributing.html).
  [... cut]
closed 2018-09-08 by wholmgren