pvlib/pvlib-python #472: Effective irradiance units

cwhanse, 2018-06-01:
`pvsystem.sapm` expected `effective_irradiance` in units of suns (1 sun = 1000 W/m2).  `pvsystem.calcparams_desoto` expects `effective_irradiance` in W/m2.  I want to change to consistent units for effective irradiance.  

Preferences for suns or W/m2?

thunderfish24, 2018-06-02:
It seems like the Desoto quantity is an actual broadband total irradiance, possibly normalized but not "effective"---[poa_global with irrad_ref in pvlib](https://github.com/pvlib/pvlib-python/blob/master/pvlib/pvsystem.py#L947) or [S or S/Sref here](https://pvpmc.sandia.gov/modeling-steps/2-dc-module-iv/diode-equivalent-circuit-models/de-soto-5-parameter-model/), while the [unitless SAPM quantity](https://pvpmc.sandia.gov/modeling-steps/2-dc-module-iv/effective-irradiance/) seems more properly called an "effective" irradiance as a ratio of currents.

thunderfish24, 2018-06-03:
Also, calling this "effective irradiance" might be confused with the G_{eff} appearing in the DeSoto's [master's thesis](http://sel.me.wisc.edu/publications/theses/desoto04.zip), which appears to have a different definition.

thunderfish24, 2018-06-03:
It appears above that I misunderstood the definition of the parameter `S` (it is "absorbed", and thus includes some losses). My underlying concern stands though, as it seems that we are struggling with making a consistent function API out of different models that construct "analogous" parameters somewhat differently. Because I am not an everyday user of some/all of DeSoto vs. CEC vs. PVsyst, I don't think I should comment on what is better: consistent API or fidelity to the original model formulation.

I will also point out that my difficulties wrapping my head around these various irradiance definitions is what lead me to yet another formulation of "effective irradiance". This formulation does not eliminate all the various issues either, but in some measurement setups it is significantly easier to reason about.

adriesse, 2018-06-03:
I would prefer things called "irradiance" to have units of W/m2.

cwhanse, 2018-06-04:
@thunderfish24 we are resolving the conflict about the irradiance input to `calcparams_desoto` as part of #462 - see my comment on this topic late in the discussion.

cwhanse, 2018-06-04:
As it is now, calcparams_desoto accepts irradiance in W/m2, sapm expects irradiance in suns.  If we agree, the change will be to sapm to accept irradiance in W/m2.

adriesse, 2018-06-04:
The original sapm report uses suns too. Maybe there is a role for an `effective_irradiance_factor` to link the two.

cwhanse, 2018-06-04:
That's an idea. If we want to retain the units published for each model, I could see implementing a private helper function to convert units based on model.  Not sure where it would plug into the model chain workflow, though.

wholmgren, 2018-06-04:
I prefer ``effective_irradiance`` in units of W/m^2. I prefer prepending or appending "normalized" for unitless quantities.

adriesse, 2018-06-06:
Here another thought out of the left corner. Perhaps these functions should have just `irradiance` as a parameter, without further details.  There is a similar situation with inverters, where it is clear there are some losses between the array and the power electronics, but the input to the inverter is just called `p_dc`, not `p_dc_after_wiring_loss`.

cwhanse, 2018-06-06:
@adriesse I think that would be a step back for pvlib in terms of our objective to clarify the component models and their usage. For the inverter example you cite, I'd be inclined to add clarity to the `p_dc` input.

adriesse, 2018-06-06:
I admit I didn't have that objective foremost in mind.

cwhanse, 2018-06-11:
For this issue, my plan is to implement `effective_irradiance` in units of W/m2 for SAPM. I believe these changes will make the use of effective irradiance consistent throughout pvlib: 
- change `pvsystem.sapm` to expect `effective_irradiance` in units of W/m2
- no change needed to `pvsystem.sapm_celltemp`
- clarify signature of `pvsystem.samp_effective_irradiance` to add units to input parameter descriptions.

PR #815 (2019-11-26): Change units on SAPM effective irradiance from suns to W/m2
 - [x] Closes  #472
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/contributing.html).
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests (usually) must pass on the TravisCI and Appveyor testing services.
 - [x] Updates entries to `docs/sphinx/source/api.rst` for API changes.
 - [x] Adds description and name entries in the appropriate `docs/sphinx/source/whatsnew` file for all changes.
 - [x] Code quality and style is suffic [... cut]
closed 2019-11-26 by cwhanse